prusa3d/PrusaSlicer #1272: Wrong windows position (autocenter)

alsopub, 2018-09-30:
### Version
Slic3r PE 1.41.1-beta

### Operating system type + version
MacOS 10.9.5

### Behavior
Previous versions remember window position and size.
New (beta) version trying to center window but do it wrong.
Here is some screenshots with different window size:

http://dl4.joxi.net/drive/2018/09/30/0003/3259/253115/15/5b5434cbcc.png
http://dl4.joxi.net/drive/2018/09/30/0003/3259/253115/15/de00d79e17.png
http://dl3.joxi.net/drive/2018/09/30/0003/3259/253115/15/408e1ade12.png

Patrik-svobodik, 2018-09-30:
WTH do you use such old OSX? But it does happen to me on Mojave and High Sierra

wabbitguy, 2018-10-01:
Some of the same thing here on (High Sierra 10.13.6) window size is correctly remembered between runs.

Position never is, window appears towards centre of the monitor and I have to drag back to top left.

wabbitguy, 2018-10-05:
@vojtechkral I just downloaded and tested the 1.41.1RC version. The window opens in the middle of the monitor (sort of). I can drag the window to the top left of the monitor, but next time I run the app it doesn't remember the new position it defaults back to the middle.

If I hold down the OPTION key and click on the green dot in the window (to expand to almost full screen) and then quit the app, next time it remembers this position and opens properly.

Resizing the window, however never gets saved.

Hence, I see the bug as a "window size/position changed" event that is not being saved when the user either repositions the window or resizes it.

Mel

Patrik-svobodik, 2018-10-05:
Sorry for off topic, @vojtechkral how far are you with the dev_native branch?

vojtechkral, 2018-10-10:
@Patrik-svobodik There's still a lot of work to be done.

I'll have a look at the window position persistence as part of dev_native, since we're using newer wxwidgets now...

bubnikv, 2019-01-28:
It will be fixed in the upcoming Slic3r 1.42.0-alpha4

closed 2019-01-28 by bubnikv