prusa3d/PrusaSlicer #1274: Stop Firmware Notifications in generated .gcode in default profiles

compmas2, 2018-09-30:
### Version
1.41.0

### Operating system type + version
Debian Linux 9.5.0
Windows 10

### Behavior
I have had to revert my Prusa i3 MK3 to the 3.3.1 firmware since the 3.4.0 has a bunch of issues with the filament sensor and is failing print jobs when printing through OctoPrint during some of these false filament detection issues.

Since I have reverted the firmware and use Slic3r 1.41.0, I am now getting a warning each time I go to print that states a new firmware is available.  This message seems to be generated by the following custom gcode that is in Slic3r by default:
`M115 U3.4.0 ; tell printer latest fw version`

My main issue with this is that I now cannot remotely start print jobs through OctoPrint with any gcode sliced through the new version of Slic3r because the printer beeps and displays a message waiting for the button on the front of the printer to be pressed.

My main request it to remove this custom gcode in the default profiles.  If you still feel this needs to be there, add a timeout to the message (30 seconds or 60 seconds) in the firmware so that a print job can continue.

bubnikv, 2018-10-01:
You can save the printer profile under your new name, where you can remove
the M115 line from the stat G-code. After a push of the profile update by
Prusa Research, you will see the changes in your copy of the printer
profile highlighted with a black color, while the Prusa recommended values
will be green.

Is this a sufficient solution to you?

compmas2, 2018-10-03:
Thank you for the information.
My only recommendation would be that the firmware notification not hold up a print job until manual intervention, but I am assuming that is a firmware request and not for Slic3r.

bubnikv, 2018-10-03:
> My only recommendation would be that the firmware notification not hold
up a print job until manual intervention, but I am assuming that is a
firmware request and not for Slic3r.

Yes, that's a request to the firmware team.

Also it makes sense to show the message on the printer for a particular
firmware update just once, then keep quite.

bubnikv, 2018-10-05:
We will not change slic3r. Please notify the firmware team
https://github.com/prusa3d/prusa-firmware/issues
if you think you need a firmware change.

Closing.

closed 2018-10-05 by bubnikv