prusa3d/PrusaSlicer #1277: Slic3r usese Nozzle diameter to decide first layer after raft thickness.

gvJaime, 2018-09-30:
### Version

1.41.0+linux64

### Operating system type + version
Fedora 28

### Behavior

#### Explanation

The first layer thickness of a rafted model is the sum of the Contact Z distance and the Nozzle diameter. I guess this was done for a reason in vanilla Slic3r, but this should be a separate setting. As it is now, it results in too thick layers and bad first layer appearance. It also happens when some other area of the model is supported

#### Reproducing

You can use a simple cube and print with a raft to observe this.
With standard settings, and using a non-soluble raft (Contact Z distance > 0, 0.1 for example), the first supported layer is a whooping 0.5mm high. 
By reducing Contact Z distance to almost 0, you can go down to 0.4, but won't go any down than this. 0.4 is still too thick.

#### Workaround

You can go to the printer .ini and change the nozzle_diameter to a false, lower value. However this may have some other unintended consequences.

Sebastianv650, 2018-09-30:
This is quite the same basic topic as #102.
It's a very old discussion, already started years ago over in the original Slic3r. As you noted, a supported extrusion is assumed as zero-squished extrusion of nozzle diameter. Some configurable squish (with adjusted extrusion value to keep the wanted line width) to that extrusion would be very helpful in terms of optical quality and helping some materials to stick.

Fingers crossed that they finaly implement such a feature during their now started support code improvements.

bubnikv, 2018-10-01:
I agree, that we need a separate settings for the 1st layer extrusion above the raft, and for the squish against the raft.

@Sebastianv650 Regarding #102, one shoe will not fit all the feet. For the raft-like supports, a raft-like settings will be perfect. For the supports on organic and / or complex shapes (toys, models, cosplay), the raft-like settings will make the supports impossible to remove.

bubnikv, 2021-02-24:
Implemented with #6003 and aee136c

Object over raft is no more printed with bridging flow, but with a normal flow.
Gap between the raft and 1st object layer is configurable independently from the support gap.

closed 2021-02-24 by bubnikv