prusa3d/PrusaSlicer #1278: Vase mode skips layer in preview and when printing

ThreeSevenths, 2018-09-30:
### Version
1.41.0+win64
1.41.1-beta+win64-201809261048

### Operating system type + version
Windows 10 Pro x64 1803 OS Build 17134.285

### Behavior
Slicing this model produces no visible errors in preview, but after sending to printer or exporting GCode, a visible layer is partially missing, and the resulting printed object contains an error in that location. The model is scaled to 30% original size.

#### STL/Config (.ZIP) where problem occurs
[slic3r_bug.zip](https://github.com/prusa3d/Slic3r/files/2431818/slic3r_bug.zip)

FidelCapo, 2018-10-01:
Thank you for report. You're right. I've found that this is caused by the combination of Vase mode and Detect thin walls settings. When I turned off Detect thin walls, I did not see any missing layers. Can you confirm this? Thank you.

ThreeSevenths, 2018-10-01:
Yes, that fixes it! Thank you very much @FidelCapo

bubnikv, 2018-10-05:
@YuSanka would you please disable the gap fill, when the vase mode is activated? Plase do it in dev_native branch.

rtyr, 2020-04-16:
Implemented. Closing.

closed 2020-04-16 by rtyr