prusa3d/PrusaSlicer #1279: Slicer cannot create G-Code files

jventerprises, 2018-10-01:
### Version
1.41.0+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 Home 64 bit

### Behavior
Open any STL
click slice
click export G-code

[image]

I have run the application as administrator
I have ensured that the folder I am working from has all permissions needed

I can see the .tmp file created,
I can see the .times file created

I can occasionally get a competed g-code on the first export on a clean run

_Is this a new feature request?_  No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[3DHubs_Marvin.zip](https://github.com/prusa3d/Slic3r/files/2432321/3DHubs_Marvin.zip)

jventerprises, 2018-10-01:
ah, it appears to be fighting with my QNAP QSYNC application.  Whenever I try to write files to a directory managed by my NAS I get this error.

Not sure this is Slic3rs problem so I am closing it

closed 2018-10-01 by jventerprises

bubnikv, 2018-10-01:
It is a known issue of renaming files on Windows, if you have some file
monitor application active. The upcoming Slic3r release will have a
workaround implemented to retry renaming a file after a bit of delay.

https://github.com/prusa3d/Slic3r/commit/9d9e4a0f7b2d9d3440c2a05bc65b2cab707d148a