prusa3d/PrusaSlicer #1280: Original Prusa bed image is only displayed if the "Origonal Prusa i3 MK2" profile was open last before launch.

bagobones, 2018-10-01:
### Version
1.41.0+win64

### Operating system type + version
Microsoft Windows [Version 10.0.17134.285]

### Behavior
Original Prusa bed image is only displayed if the "Origonal Prusa i3 MK2 profile was open last before launch.

Saving an inherited profile and launching with any of those inherited profiles will revert to the standard non image based Slic3r bed grid  

#### STL/Config (.ZIP) where problem occurs
No STL needed, this issue happens based on profile and launch.

bubnikv, 2018-10-01:
We were not able to reproduce your issue. We had this issue before in some 1.41.0 beta versions, but not in the 1.41.0 final. Are you running the 1.41.0 final? Would you please try to describe your steps to reproduce the issue in more detail, if it is possible?

bagobones, 2018-10-02:
I am using the ORIGINAL PRUSA I3 MK2S and the DRIVERS & APPS 2.2.0 installer, I upgraded from the previous release also using the DRIVERS & APPS installer https://www.prusa3d.com/drivers/

I could uninstall everything and try deleting all the user profiles but I would prefer not to lose my presets.

bubnikv, 2018-10-05:
Do you still have the issue?

Would you please zip your Slic3rPE directory and send it to us?
You will find the Slic3rPE on Windows in

"C:\Users\your_user_name\AppData\Roaming\Slic3rPE" 

or 

"C:\Documents and Settings\your_user_name\Application Data\Slic3rPE"

bagobones, 2018-10-08:
I scrubbed my Octoprint API Key from the printer configs and snapshots.
Not the most exciting config.
[Slic3rPE.zip](https://github.com/prusa3d/Slic3r/files/2454611/Slic3rPE.zip)

FidelCapo, 2018-10-08:
I've found that our Slic3r doesn't always display the texture when switching between profiles correctly. I think one of the reasons may be that your profiles aren't inherited from the system profile (they are displayed as profiles inherited from the default profile in which the Prusa bed texture should not be displayed). From which profile is the profile inherited, is displayed in Dependencies.

When I created a new profile inherited from the system Original Prusa i3 MK2 profile and set into your settings, the texture was displayed correctly (select the MK2 system profile, enter settings changes, and save).

Can you try create new profile inherited from system profile if it works correctly? Thank you.

bagobones, 2018-10-08:
Maybe there was a bug in 1.40.X when these profiles where created.

I deleted and recreated my main profile AGAIN from the original profile.. looks like the dependency is sticking.

closed 2018-10-08 by bagobones