prusa3d/PrusaSlicer #1282: Feature request: Move purging volumes to Filaments Tab

3d-gussner, 2018-10-02:
### Version
1.41.1-beta+win64

### Operating system type + version
Win64

### Issue:
Changing filaments for different mm parts is quite common as we don't want all of them in only this 5 colors/materials. At this moment you have to adjust the purging volumes for each combination of your filaments depending in which mm extruder they are placed.

Having the settings in the Plater Tab it makes it impossible to save these as separate settings. Or i couldn't find it.

### Suggestion:

- Wouldn't it make more sense to add MM purge values to the filament itself and not to the extruder, as the needed purge volume depends on the filament?
 
- As the bleeding into new filament is quite complex and is depending on the combination of those two materials the "purging volume adjustment" would be still great for fine adjustments.

- Move "purging volume adjustment" under Printer Settings Tab so we can save these settings.

lukasmatena, 2018-10-02:
> ... as the needed purge volume depends on the filament?
Not only that. It depends on filament and its color, so changing it in filament settings would require having a separate value for each color. Also, it strongly depends on the previous filament that was just unloaded, so it would not be enough to save it in filament settings. You actually answered yourself in the second suggestion.

3d-gussner, 2018-10-03:
@lukasmatena 
I am sure you have discussed it what the best possible options are, but as the "Purging volume adjustment" windows explains "Volume to purge (mm³) when the filament is being ... unloaded ... loaded".
Never less I still think it makes much more sense to have these values in within the filaments, as the "Show advanced settings" window calculates the "basic" settings to get the right change pattern.

The idea having a 2nd value (multiplier/factor) for the amount of pigments could make sense.

Here my 1st thoughts on that
**Example:**
Let's say the 50mm³ for unload and load is the minimum you should have to prevent bleeding.

- 1st Filament is translucent pure material, which means it has no/few pigments that could bleed into another one.
This could have a pigment factor of 0.1 and an unload/load volume value of 50/50

- 2nd Filament is black which has on the other side has much more pigments which will bleed especially into lighter colors.
This could have a pigment factor of 1.5 and an unload/load volume value of 120/50

- 3rd Filament is yellow and has normal pigments which could bleed into other one.
This could have a pigment factor of 1 and an unload/load volume value of 70/50

At this moment the change pattern would be (this calculation is made from the basic settings):
1st to 2nd 100 mm³ = 50+50
1st to 3rd 100 mm³ = 50+50
2nd to 1st 170 mm³ = 120+50
2nd to 3rd 170 mm³ = 120+50
3rd to 1st 120 mm³ = 70+50
3rd to 2nd 120 mm³ = 70+50

With the pigment factor:
1st to 2nd 80 mm³ = 50*0.1+50*1.5
1st to 3rd 55 mm³ = 50*0.1+50*1
2nd to 1st 185 mm³ = 120*1.5+50*0.1
2nd to 3rd 230 mm³ = 120*1.5+50*1
3rd to 1st 75 mm³ = 70*1+50*0.1
3rd to 2nd 145 mm³ = 70*1+50*1.5

In my opinion this reflects more the experience that others had and shared. See Chris Warkocki
 video "Slic3r/MMU2 Bleed control" https://www.youtube.com/watch?v=KHPLS0apvG4 

I think that would make the use of the "Show advanced settings" less often. Maybe a different approach using percents in this tab could be used to automatically adjust the pigment factor depending of the values users find being right during their tests.

Users could share their values depending on their used filaments and change pattern combinations. this could be easy accumulated into the Slic3r-settings.

> I don't think it makes any sense to save it under Printer Settings.

You are write it is my bad I mean the "Print Settings" as here we have the "Wipe tower" part and not "Printer Settings". Sorry for the typo.

> However, the values should be saved during export of AMF/3MF.
This would mean i need to have lot of AMF/3MF settings saved even for just 5 colors if they are orientated in different ways.

Having the unload/load and pigment factor in the filament settings would make the change pattern automatically independent if you put one filament in tool 1 or another one.
Using the "advanced" window like i described above, where this would automatically adjust the related filament settings, could make the need of it less and less. As the filament pigment factor would get dialed in more and more accurate every time. That is at least my theory.

lukasmatena, 2018-10-05:
@3d-gussner 
Thanks for you suggestions. I don't quite understand the purpose of the 'pigment factor', which seems to overcomplicate the inputs while adding nothing new. It always multiplies the load/unload values, which are also configurable, so unless I misunderstood what you wrote, the pigment factor is totally redundant.

Anyway, you basically suggest that the dialog stays as it is, but the default values (loaded/unloaded) are configurable in filament settings, am I correct? This still does not solve the issue I mentioned above, that you'd need a different value for each filament colour, so the user would have to derive a custom profile for each of the spools he has and keep track of them (which is quite possible to do with current profile management). Have I understood correctly?

I don't believe that storing the two values in filament settings will be universal enough, I fear that if someone finds the "optimum" based on his set of e.g. 5 filaments, he will find that the values are not optimal when he adds a sixth filament in the group, and that he can't get the "best" outcome for all the additional combinations (there are 10 of them) with only two additional parameters to change. But this may be an academic problem and the difference may actually turn out to be insignificant. This was also the reason why the "advanced" settings were introduced.

**Is there someone else with experience with MM printing (!!!) and some constructive opinion on this?**

bubnikv, 2018-10-05:
We will likely extend the filament settings with parameter groups depending on filament color (or flavor / brand if you wish) one day, so there will be an option to set the wipe values at the filament level. But it certainly not be this year.

@3d-gussner Internally, I was proposing to implement an "opacity" parameter, by which the software would fill in the wipe matrix. I understand this is a similar concept to your "pigment amount". One has to try various formulas for constructing the wipe matrix from opacities over a large number of filaments and prints. This could be done without changing Slic3r, so a volunteer is needed, and I am marking this issue as such.

mdjurfeldt, 2019-02-17:
Not sure if this is the best place to write this, but it has from the start been obvious to me that printing with PLA or PETG together with BVOH or PVA has a devastating effect on layer adhesion with the standard purge settings. I think this is a very strong argument for filament specific purge settings. For BVOH, I always set unloading to 170 mm3 giving a total purge of 240 mm3 (for the standard 70 mm3 load setting). I've confirmed that any value under 170 mm3 gives weaker layer adhesion. For PVA, the unload value needs to be set even higher.

3d-gussner, 2024-03-21:
Closing as PS 2.7.3-alpha1 should be similar to my proposal.

closed 2024-03-21 by 3d-gussner