prusa3d/PrusaSlicer #1284: Bridging section of perimeters starts in midair.

EvilGTV, 2018-10-03:
I've had this bridging problem I think for a long time (IE. regardless of Slic3r version, it certainly happens in vanilla Slic3r as well), which is that although Slic3r detects the bridge just fine, it doesn't apply bridging parameters to the perimeters until AFTER it's already on the bridge.  I have spent hours trying to tune my bridging every time I try a new filament, and always have problems with the perimeters behaving completely differently to the infill (usually hanging looser, and not really changing much regardless of settings).  Trying to print the last day or two in TPE, bridging has just not succeeded at all - no matter what I changed and how far, could NOT get the start of the perimeters to stick in the least bit.

Doesn't really matter what bridge it is, any of the bridge tests I've opened show this problem, along with a phone case yesterday.  The image below should illustrate the problem clearly - you can see that the bridging speed change doesn't occur in the 3 perimeters until the nozzle is already printing in thin air.  Also note that the bridge infill is overlapping the pillar just fine.

[image]

For an experiment, I first tried to print as it was output, which failed horribly (as it had the last 20 or so times I'd tried with various wildly different parameters, including turning 'detect bridging perimeters' off) - the perimeters didn't anchor at the start, then the loose ends got blown and dragged around, and interfered with the infill section, leading to complete failure - a blobby mess between two pillars.  I then opened up the G-code, and manually edited the point at which it changed from using 'perimeter' settings to 'bridging' settings, to give the result below.  I simply aligned the change point with the ends of the infill section.

[image]

This did exactly what I expected to happen - the perimeters anchored perfectly, and I had a very serviceable and successful 50mm TPE bridge with just a few loose strands, one that I could actually tune to get an excellent result instead of just a different shaped mess.  So at least I know how to fix it, but manually editing the G-code like that every single time I want to change a variable when tuning bridges?  No thanks....

The obvious and quickest short term solution is to do exactly what I've done here in the code - align the change point for perimeters with the bridging infill ends, but I do wonder if in the future, there might be scope to add user editable parameters for changing bridging overlap for both perimeters and infill separately?  Maybe even some editable detection parameters, similar to when generating support?  Could allow for another dimension in really anchoring down some of the trickier filaments?

Using 1.41.1-Beta-Win64 and Windows 8.1, but as I mentioned, this behaviour occurs with regular Slic3r as well.  Although of some interest is that regular Slic3r seems to change over at least slightly closer to the pillars, but still in thin air.  I also note that although this issue: https://github.com/prusa3d/Slic3r/issues/1185 was about the infill not attaching to the pillars, the third photo down in the first post shows the same behaviour in the perimeters as I describe.

[config and stl.zip](https://github.com/prusa3d/Slic3r/files/2441051/config.and.stl.zip)

EvilGTV, 2018-10-04:
I've been doing a little experimenting, and I believe there is potential to greatly improve printing results by adjusting (or even better, giving options to tune) the detection/application of bridging perimeters.  Although I'm not sure that the perimeters aren't being treated by the overhang detection, seperate from bridging infill.

The below is a slice of a flexible phone case I've been experimenting with - I assumed it would need support just by the nature of it's right angle overhang, but I saw that some had printed it without support, so I cut a segment out to try.  In the image, I have changed in G-code the point at which bridging parameters are applied to the outer perimeter, moving it back 1mm so that it changes while on 'solid ground'.  Previous to my edit, the change point lined up with the small purple bit you can just make out.

[image]

And below is the two prints - unmodifed G-code on the left, and my modification on the right.  You can immediately see that the unmodified version mostly dropped that outer strand, while simply changing over to bridging parameters 1mm earlier meant that the outer strand anchored to the previous one across its length.  Of interest in the G-code is that this previous strand is detected as an overhang (I assume) because the fan speed is ramped up to 100%, but it doesn't apply any bridging parameters, even though it too is basically printed on nothing.  I suspect changing this perimeter to also print using bridging settings may improve the result even further.

[image]

What I'm wondering is whether 'bridging' perimeters as in my first post, and 'overhang' perimeters as in this one are treated differently, or the same?  I suspect the latter?  In which case, perhaps there is some advantage to separating them out, as if someone was using a high bridging speed to span a gap, it seems this would be detrimental to a severe overhang case like this one where you really want time for the filament to fuse to the previous perimeter.  

Perhaps adding an 'overhang' speed in the speeds box, and some user configurable overhang detection and bridge detection options - along with a possibility to set the 'overlap' with existing material for each of those?  I know, becoming more of a feature request than a bugfix, but in my opinion those could be considered for the long term, while in the short term more aggressive application of bridging parameters to overhangs/bridging is hopefully the 'easy' workaround.

supermerill, 2018-10-04:
> What I'm wondering is whether 'bridging' perimeters as in my first post, and 'overhang' perimeters as in this one are treated differently, or the same?

same

EvilGTV, 2018-10-04:
Hopefully that works for you...  I haven't worked out how to save an STL from Fusion yet, so it's been output to Repetier and saved from there.

[phoneslice.zip](https://github.com/prusa3d/Slic3r/files/2445440/phoneslice.zip)

EvilGTV, 2018-10-10:
No, it is not.  If you actually read my first post, you'll see I take great care to distinguish between bridging infill and bridging perimeters, as the infill section of the bridge works as it should, and the perimeters are fixed by changing them to behave like the infill.  In fact the last line before the first photo is "Also note that the bridge infill is overlapping the pillar just fine." 

Even more than that, you would have seen the previous issue I linked to, where the issue with 0% infill is discussed, and a fix has already been created...  Which looks identical to your post.  Incidentally, have a look at where your perimeters change settings in your picture with 5% infill....

4nthonylin, 2020-07-24:
Any updates on this? I have noticed this as well and it looks like the change in speed causes the nozzle to pull the strand thin potentially breaking the strand. This causes the strand to fall or have an uneven extrusion.

Grawp, 2022-01-09:
> I have noticed this as well and it looks like the change in speed causes the nozzle to pull the strand thin potentially breaking the strand. This causes the strand to fall or have an uneven extrusion.

I too am experiencing this.

Model and starting settings: (material is Fiberology PCTG)
[bridge.zip](https://github.com/prusa3d/PrusaSlicer/files/7834852/bridge.zip)
[image]

Here are my experiments with better cooling, disabling LA, disabling retractions, slowing down speed. You can look up specific settings in comments inside gcodes:
[experiments.zip](https://github.com/prusa3d/PrusaSlicer/files/7834856/experiments.zip)

closed 2026-09-29 by prusaslicer-bot[bot]