prusa3d/PrusaSlicer #1285: directory problem

bjku96, 2018-10-03:
### Version
Slic3r Prusa Edition 1.39.1

### Operating system type + version
Windows 10 64bit

### Behavior
Hello, when I want to start Slic3r with the slic3r.exe program, i get a black window and nothing happens. 
So I wrote a batch file with the following content:

@perl5.24.0.exe slic3r.pl %*
PAUSE

and got following result: 

[image]

I thought that it would be the exponent  ³   in the path but copying the slic3r-folder to C: did not solve the problem and I never had issues with Slic3r located on the desktop.

I solved it by editing the batch to: 

@perl5.24.0.exe slic3r.pl --datadir data%*
PAUSE

Now it starts, but I don't like that solution and I have the same problem with the current version of slic3r .

bubnikv, 2018-10-03:
This looks similar to https://github.com/prusa3d/Slic3r/issues/1271

We did not do any change on the slic3r.exe wrapper, but it may be, that Microsoft changed something behind the scenes on the command line processing?

What Windows 10 flavor are you running on?

bjku96, 2018-10-04:
I'm using Windows 10 Education, version 1803.

bubnikv, 2018-10-05:
By the way, Slic3r PE  1.39.1 is very old. Why did you not upgrade?

If Slic3r hangs on start with the default datadir, would it help if you delete it?
On Windows, you should find it in

"C:\Users\your_user_name\AppData\Roaming\Slic3rPE" 

or 

"C:\Documents and Settings\your_user_name\Application Data\Slic3rPE"

bubnikv, 2018-10-11:
Closing then. What a pity, that we did not get the content of your broken Slic3rPE folder.

closed 2018-10-11 by bubnikv