prusa3d/PrusaSlicer #1288: Missed bridging layer

sigxcpu76, 2018-10-04:
### Version
1.41.1-beta+
### Operating system type + version
MacOS

### Behavior
A bridging layer is completely missing.
I have attached the original .STL, generated G-Code and a preview picture that best describes the problem.

#### STL/Config (.ZIP) where problem occurs
[missing_layer.zip](https://github.com/prusa3d/Slic3r/files/2448394/missing_layer.zip)
[image]

sigxcpu76, 2018-10-05:
If it helps, tried the same file with 1.41.0 and it doesn't have this issue.

bubnikv, 2018-10-05:
This is an unfortunate regression error, which is already fixed on the vb_slicing_fix2 branch. It will be released hopefully today as a 1.41.1-rc.
Closing.

closed 2018-10-05 by bubnikv