prusa3d/PrusaSlicer #1289: Slic3r 1.41.1-beta - Prusa Edition crashes

AndylgTom, 2018-10-05:
Version
Slic3r 1.41.1-beta

Operating system type
Windows 10 Build

Behavior
Describe the problem

The program after re-changing the language and trying to open the model - program crashes. (Does not depend on language and model)

Video crashes - https://yadi.sk/i/sXyAUe5jOfgU7w

Atatoth, 2018-10-05:
In my case program crashes when printing few times and I back to 1.41

bubnikv, 2018-10-05:
> In my case program crashes when printing few times and I back to 1.41

I don't understand the statement: "and I back to 1.41"
Do you mean 1.41.0 works for you correctly?

AndylgTom, 2018-10-05:
I figured it out. I do not know what exactly the problem was. I just deleted all the settings in Roaming\Slic3rPE and now everything is fine.
Close.

closed 2018-10-05 by AndylgTom

Atatoth, 2018-10-05:
@bubnikv  yes in 1.41 all prints from USB are ok no crashes. I have Windows 10

bubnikv, 2018-10-05:
@Atatoth 
> In my case program crashes when printing few times and I back to 1.41
> yes in 1.41 all prints from USB are ok no crashes. I have Windows 10

I still do not understand your problem / question / issue / request. Would you please be more specific?