prusa3d/PrusaSlicer #1290: Feature Request: Machine Limits beign optional

FelixHauser, 2018-10-06:
### Version
Slic3r PE 1.41.1 rc

### Operating system type + version
macOS (or any other for that matter)

### Behavior
I'm not sure if the "Machine Limits" sub-section on the "Printer" section is new to this release or some of the previous I skipped, but here is my request:
When I updated Slic3r some values were already set up for me in this sub-section, which might or might not be suited to my machines. Although I appreciate and I will make intensive use of this new sub-section, I would like it to be an optional thing. I believe that a checkbox with something like: "use firmware values" or something of the sort would be extremely useful. For instance one might have a printer without an open source firmware or one could have no idea of where to get the values from.

Thanks in advance

FH

Khenals, 2018-10-09:
I need that checkbox pls.

Braccoz, 2018-10-10:
i second this request. it is utterly stupid to have to manage machine limit in both the firmware and the slicer. i want to be able to turn it off

bubnikv, 2018-10-11:
Duplicate of https://github.com/prusa3d/Slic3r/issues/1212

> it is utterly stupid to have to manage machine limit in both the firmware and the slicer. 

It is utterly stupid to have the Slic3r operate with different values than the firmware.

Closing.

closed 2018-10-11 by bubnikv