prusa3d/PrusaSlicer #1293: Very different speeds on layers

Hywelmartin, 2018-10-07:
### Version
1.41.1-beta-linux64

It has something to do with "slow down if layertime is lower than xx" and if speed are high...

[image]

### Operating system type + version
ubuntu 18.04

### Behavior
* Very different speeds on layers that would have had almost the same speed.. in my opinion at least

* slice and study the preview
 

[image]
[image]
[Dorus.zip](https://github.com/prusa3d/Slic3r/files/2453800/Dorus.zip)

FidelCapo, 2018-10-08:
Can you please send a 3MF or config file? Thank you

Hywelmartin, 2018-10-08:
here you go..

bubnikv, 2018-10-11:
I spent around half an hour investigating this issue and I am not quite sure whether there is a bug at all, or whether there is a bug in the visualization, or whether there is a bug in the slow down logic.

The following visualization comes from Simplify3D.

[image]

This visualization comes from Slic3r.

[image]

Now we can see, that the colors are much sharper on Slic3r. @enricoturri1966 , would you please look into the visualization, whether the colors are assigned correctly?

Panayiotis-git, 2018-10-11:
I think that S3D uses a lot of greenish colors. So, a speed difference that would change color in Slic3r, in S3D it would only change the green hue. Otherwise they look the same.

Hywelmartin, 2018-10-11:
@bubnikv the slower layer across the eyes and shoulders is noticeable in the printed part...

rtyr, 2020-05-24:
Seems OK to me in the current 2.2.0. Speed changes are corresponding to cooling slow down logic. Closing.

closed 2020-05-24 by rtyr

Hywelmartin, 2020-05-28:
[image]