prusa3d/PrusaSlicer #1294: Feature: Slic3r should remember the serial port and baud rate

timur-tabi, 2018-10-07:
Slic3r defaults to 2500000 baud for any USB connection on my Mac, but that speed never works for me.  I don't know if it's my Mac or the MK3 that can't handle it, but every time I connect I need to change the baud rate to 115200.

It's highly unlikely that the serial port or baud rate will change from one day to the next, so Slic3r should do a better job of remembering it.  I have to set the USB port and baud rate every time I launch Slic3r.

bubnikv, 2018-10-08:
Why don't you save the serial port speed into the printer profile?

timur-tabi, 2018-10-09:
Hmmm.... I guess that works.

Is 250000 baud supposed to work?  I can't get anything about 115200.  I'm about to install Octoprint, so it will be a moot point soon, but it seems that Slic3r really should default to 115200 baud since it's more reliable.

bubnikv, 2018-10-11:
The serial speed is often hard coded into your printer firmware, 250 000 is just fine.
Closing this issue.

closed 2018-10-11 by bubnikv