prusa3d/PrusaSlicer #1296: Legend box Issue

adegidio, 2018-10-08:
### Version
_Version of Slic3r_ 1.40.1+win64

### Operating system type + version
Windows 10 64bits

### Behavior
* I can't see scale legend correctly, it looks like a simple box grey colored
* Always I press "_Slice now_" I get the piece sliced. Then I press "_Export G-code..._" and save the file. After saving the file I have a default view of _Feature type_ and the **Legend Box** looks like a grey colored one without any information inside. This problem is with all views (Feature type, Height, Speed, etc)
* _Expected Results_: I expected to see the information of every color used in view after saving the file.
* _Actual Results_: I just see a grey box in upper left corner.

[image]

_Is this a new feature request?_ No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[config.zip](https://github.com/prusa3d/Slic3r/files/2456837/config.zip)

bubnikv, 2018-10-11:
Currently the legend is only shown on Windows with OpenGL 2.0 and newer. 
@enricoturri1966  is working on a fix for the next release.

bubnikv, 2018-12-12:
We are deprecating OpenGL older than 2.0 in Slic3r 1.42 alpha. On Windows, our binaries will be shipped with a software renderer library MESA to emulate OpenGL 2.0 on old computers.
This issue will be gone on Slic3r 1.42 alpha, but the MESA software renderer will likely be slower than the default OpenGL 1.4 renderer by Microsoft. We recommend you to update the graphics driver on your system.

Closing.

closed 2018-12-12 by bubnikv

adegidio, 2018-12-13:
Hello, I've installed OpenGL Extensions Viewer.
As I can see I'm running OpenGL 3.1:
[image]

bubnikv, 2018-12-13:
well, let's hope for the best with the 1.42 alpha update. It compiles
against an updated wxWidgets library, so it may detect OpenGL correctly.