prusa3d/PrusaSlicer #1297: Small perimeters setting customization

dmitry-kutergin, 2018-10-09:
### Version
1.41.1

### Operating system type + version
Ubuntu 18.04

### Behavior
I have a question regarding "Small perimeters" speed settings. A tooltip says it is applied to the perimeters with a radius of <= 6.5 mm. 

- If the perimeter is irregular shape how this radius is calculated?
- Is it possible to make this radius of 6.5 mm configurable somewhere?

ssavickiy, 2021-11-03:
I also think that a changeable setting of small perimeters is necessary

bubnikv, 2021-11-03:
The perimeter is considered small if its perimeter length is 2*PI*6.5mm.

viper0078, 2021-11-21:
I also had trouble with the explanation of this parameter.
It's not the radius, but the length of the perimeter.
For a shape like the one shown below, this parameter is not necessary.
[image]

I would like to have a parameter to turn this parameter on and off.

BMacours, 2023-01-22:
+1 for this one

evilC, 2024-07-16:
cmon guys, we have waited over 5 years for this. It can't be much work, please implement!
6.5mm is sometimes too much, esp if you are trying to print text. We need to be able to configure it

BMacours, 2026-09-22:
Still very useful I think. See also #7329