prusa3d/PrusaSlicer #1298: Windows desktop.ini is not valid

drdelaney, 2018-10-09:
### Version
Slic3rPE-1.41.1-rc+win64-full-201810051336, Slic3rPE-1.41.0+win64-full-201809010758

### Operating system type + version
Windows 10, 1803

### Behavior
The software is trying to import desktop.ini files and erroring out upon start up. It continues after the error without issues (seemingly).

My desktop.ini files are added by googles backup sync, as it changes the folder icon overlay.

I am not sure if ignoring this file is an acceptable solution, as someone could set up a valid profile called desktop if desired.

Its full contents are:
[.ShellClassInfo]
InfoTip=This folder is shared online.
IconFile=C:\Program Files\Google\Drive\googledrivesync.exe
IconIndex=16

#### STL/Config (.ZIP) where problem occurs
N/A

vojtechkral, 2018-10-10:
Do I understand correctly that the Google backup sync is adding the `desktop.ini` files inside the Slic3r filament, printer, print or profiles directories?

KD7LRJ, 2018-10-10:
I am having the same exact thing on my system using Google's Drive File Stream product. Yes, it creates the desktop.ini file in each of the settings folders. If I delete them manually, they eventually come back. I just ignore the error when starting Slic3rPE. I haven't noticed any harm other than the warning message.

drdelaney, 2018-10-10:
@vojtechkral yes, it creates a desktop.ini file in all folders that it syncs.  This isn't limited to just google, but anything that creates a desktop.ini file (IE OneDrive). 
Original slic3r and older slic3rPE versions just included a generic "desktop" profile in the past, it silently accepted it. (I never understood where they came from until now, oddly enough, thought it was a default profile).

It should be safe to just ignore the desktop.ini, as it is a windows "reserved" file, and no one should be using it for slic3r profiles.

vojtechkral, 2018-10-11:
@KD7LRJ @drdelaney Thanks for the info. Out of curiosity, is the file marked as hidden/system?

drdelaney, 2018-10-11:
@vojtechkral I do not have the machine booted at this time, but the OneDrive files (tested at work earlier today) have desktop.ini marked hidden. I suspect Google's files are as well.

Patrik-svobodik, 2018-10-14:
Yeah, that could be a possible fix, ignoring hidden files

bubnikv, 2018-12-12:
Implemented with 31dc6491901c64b19fab1b6db1b3836d3da4f7ad
Closing.

closed 2018-12-12 by bubnikv