prusa3d/PrusaSlicer #1299: FR: STL File Name Placeholders in GCODE comments

kickhad, 2018-10-09:
Slic3rPE 1.41.0

Window 10

New feature request - Placeholders to add Custom GCode to include Mesh File names in GCODE comment. 

[image]

bubnikv, 2018-10-11:
I don't understand.

bubnikv, 2018-12-12:
No response, closing.

closed 2018-12-12 by bubnikv

MadeforMaking, 2020-11-14:
I think the idea is to have a placeholder for the names of the imported mesh files.

If I'm correct there is no placeholder for body_names found in the [placeholders](https://github.com/prusa3d/PrusaSlicer/wiki/List-of-all-placeholders-taken-from-the-code) doc.

I have thought about this and for me this would be useful especially when:
- I have 1 body per print, I can easily change the reference to the body inside prusaslicer instead of file save as.
- If I have multiple bodies but only have one with the object printable parameter: true, that be the body_names place holder.
    - don't need to overwrite file just to change the gcode reference
- Maybe have the body_names place holder output the list of bodies. Then you would know exactly whats inside without needing to investigate further. Not sure if this is feasible