prusa3d/PrusaSlicer #1300: FEATURE REQUEST: Ramming temperature different than print temperature

pleppik, 2018-10-09:
After a few weeks of working with the MMU2, it's clear that there are some cases where the optimal print temperature won't give a good tip on unloading, and a temperature optimized for ramming won't print well.

For example: For one TPU I'm using, if I set the print temperature low enough to prevent stringing on unloading, I don't get good adhesion between the layers. Also: When trying to use a cheap PVA for soluble support, the PLA won't adhere to the PVA unless I print the PLA too hot for reliable unloading.

This can be solved by allowing you to specify a ramming temperature different than the print temperature. There's already enough time for the extruder to heat/cool by 5-10C during the moves before the actual ramming, and more dramatic temperature changes can be accommodated by pausing a few seconds if needed.

I would suggest that the ramming temperature be set immediately before the extruder moves to the purge tower. Once the ramming sequence is complete, the temperature would be reset to the print temperature for the incoming filament (as it already is).

mylife4aiurr, 2018-10-22:
This feature is already past in kisslicer. It's needed for multiple material printing in slic3r also.

clkandcjk, 2018-10-22:
+1 on this 

I am currently using Paul’s script to post process my gcode and I have been able to print 4 filaments that were previously unprintable 

This is a must have for the MMU

Flav1u, 2018-10-25:
We need that feature !

domesticatedviking, 2019-02-20:
I wholeheartedly agree that this feature should be added.  I would say that "toolchange temperature" should be added as a field to the advanced settings on the filament profile, and that the toolchange sequence should add the necessary logic to pause while the nozzle temperature drops.

Zephrant, 2019-02-20:
Please consider this, I think it will be critical for the PETG I need to run.

domesticatedviking, 2019-03-01:
I have been trying to quantify the effects of the various advanced filament settings relating to MMU tip formation.  This chart is based on an average tip length of 5 different types of ABS. 
[image]

domesticatedviking, 2019-03-08:
As an interim solution I've begun to put this project together.   https://github.com/domesticatedviking/skinnydip

Flav1u, 2019-03-18:
Nice work domesticatedviking

tsamisa, 2019-07-28:
+1 on this issue. No matter what i do string tips are lousy.

mbordoni, 2020-10-25:
This is a must have also because today it is even worse:
Just before ramming, the printer sets the new filament nozzle temp instead of a the specific actual filament temp.
This makes the use of soluble filament or even different material switch just unfeasible since the mmu is going to get stuck at every filament switch.
Please fix this or (even better) let the user to completely manage the temp in prusaslicer by adding more custom gcode insertion in printer config and eliminating current implementation.
To do the correct filament tip a specific temperature is essential.

FEsmondeWhite, 2021-02-15:
This requested feature would be extremely helpful, I think that this would make the MMU2 much less prone to getting stuck. I would like to see an option to adjust filament temperatures specifically for ramming/retracting on filament change (eg, run at 200C for ramming/retraction and 215C for printing).

It would be great to see implemented. I'm working with PLA and MMU2. When I lower the print temp to 205, the strings I get are usually ok (gets stuck 2-3 times in 55 filament changes), but then the layers aren't well bonded. When I increase print temp to 210, then strings happen around 5x as often.

domesticatedviking, 2021-02-18:
For what it's worth @FEsmondeWhite  this feature is already implemented in Superslicer, which tracks PrusaSlicer pretty closely in terms of feature set.  [](https://github.com/supermerill/SuperSlicer/releases/).   Check out "toolchange temperature" in the advanced filament settings.

webeule, 2024-12-22:
Is there any reason why this probably simple parameter still isn't implemented?

closed 2026-09-29 by prusaslicer-bot[bot]