prusa3d/PrusaSlicer #1302: Feature Request: option to unload filament after print

timur-tabi, 2018-10-10:
I'd like a checkbox added to the UI that, when selected, adds an "unload filament" command (M702 I think) at the end of the print, before the "M104 S0".

timur-tabi, 2018-10-11:
No, it isn't.

I don't want to unload the filament every time I print.  What I want is a convenient way to say, "Ok, this is my last print with this filament, so to avoid having to reheat the hot end, just unload this filament when you're done, thanks."

supermerill, 2018-10-11:
So it's more a firmware request than a slicer one?

timur-tabi, 2018-10-11:
No, it'd a slic3r request.  I'd like to see an option the UI, perhaps next to the "Brim:" checkout, like this:

[image]

If selected, then Slic3r will add an M702 command (or the E equivalent) into the end g-code for me.

p3ck, 2018-10-11:
I personally don't think this is a great idea.  Unloading requires you be
at the machine to pull the filiment at the right time.  You are free to add
the g-code to your end-code but to add an option to the GUI endorses it and
I think it could introduce more problems.

timur-tabi, 2018-10-11:
Fair enough, but why do I need to be at the machine to pull the filament?  It's just "G1 E-100" or something like that, isn't it?

bubnikv, 2018-12-12:
I think the general conclusion is, that this feature is not needed. Closing.

closed 2018-12-12 by bubnikv