prusa3d/PrusaSlicer #1303: broken gcode generation/crashing with Slic3rPE-1.41.1-beta+linux64-full-201809261049.AppImage

jhoblitt, 2018-10-10:
### Version
`./Slic3rPE-1.41.1-beta+linux64-full-201809261049.AppImage`

### Operating system type + version
Fedora 28

### Behavior
* 1st attempt (to use the beta version at all) to slice multiple objects (1x model a, 2x model b) resulted in the bottom 8-9 layers being emitted from the generated gcode -- this wasn't noticed until 15mins into a print.  `slic3r` crashed while attempting to view the `layers` tab.
* 2nd attempt resulted in 1x model a only appearing in the gcode. `slic3r` crashes a few moments after inspecting the `layers` tab.
* Where ''crash' means the gui becomes completely unresponsive and has to be killed.

#### STL/Config (.ZIP) where problem occurs
Generated gcode from 2nd attempt, the code from the 1st attempt was lost:  https://www.dropbox.com/s/qi3cymfn4lrli06/sh_base_p1_v1_r1_0.15mm_PET_MK3.gcode?dl=0

jhoblitt, 2018-10-10:
Eh, looks like I partially lied, the gcode from the second attempt does have all 3 objects in it.

FidelCapo, 2018-10-10:
Can you please send me a 3MF file? Thank you

bubnikv, 2018-12-12:
There was no response, closing.

closed 2018-12-12 by bubnikv