prusa3d/PrusaSlicer #1307: Spiral vase start with an excessive extrusion rate

liftbag, 2018-10-11:
### Version
_1.41.1-rc+_

### Operating system type + version
_macOS Mojave Version 10.14 #s_

### Behavior
* _The first extrusion loop manifests an abundant over-extrusion at the start which decreases until it disappears at the end of the loop._
* _Any object made in spiral vase mode can reproduce the issue_
* _The first layer should not have over extrusions_
During the first loop of the spiral, the height of the nozzle relative to the plane, or to the underlying layer (those of the bottom full infill), is minimal at the beginning, and reaches the nominal value at the end of the loop. But the extrusion rate is fixed. This involves an obvious over-extrusion in the initial area of the loop, as showed in the attached video (youtube link).
* _Actually, the first layer starts over extruded and close the loop with the correct extrusion rate_
_[image]_
_https://www.youtube.com/watch?v=sgMcxs-gj9A&feature=youtu.be_

#### STL/Config (.ZIP) where problem occurs
Any object.
_[Spiral vase.zip](https://github.com/prusa3d/Slic3r/files/2467768/Spiral.vase.zip)_

FidelCapo, 2018-10-11:
I think the problem may be in extrusion width values that are set to 0 (auto). This auto setup is not well tested and we plan to remove it to future versions.
Why do you use this auto setting?

liftbag, 2018-10-14:
>This auto setup is not well tested and we plan to remove it to future versions.
> Why do you use this auto setting?
I'll report the results as soon as I can try with well-defined extrusion width values.

bubnikv, 2018-12-12:
The issue was caused by the auto extrusion width. No further response, closing.

closed 2018-12-12 by bubnikv