prusa3d/PrusaSlicer #1310: [Feature Request] Enable Wipe Tower with Repetier-Firmware

Braccoz, 2018-10-11:
### Version
1.41.0

### Operating system type + version
Windows 7 X64

### Behavior
Wipe Tower is only compatible with Marlin and RepRap/Sprinter firmware settings, selecting Repetier will throw the error

"<Slic3rPE> Configuration is not valid: The Wipe Tower is currently only supported for the Marlin and RepRap/Sprinter G-code flavors."

i don't see any reason why the wipe tower would not be compatible with repetier fw. It supports relative extruder moves.

bubnikv, 2018-10-11:
We did not test with the Repetier firmware, therefore we did not allow that. There is some hard-coded G-code generated by the wipe tower generator, and we did not evaluate whether this G-code is compatible with Repetier.

We may consider to provide you with a test build first for you.

Braccoz, 2018-10-11:
i will gladly test that out for you

Braccoz, 2018-10-24:
by the way, i tried printing with prime tower using RepRap flavor on a printer with repetier-fw and it completed without errors.

bubnikv, 2018-10-25:
ok, we will enable that in the next release (1.42).

bubnikv, 2018-12-12:
Implemented with b856775e188da6d31bc7e53926d6f3248bec85b0
Closing.

closed 2018-12-12 by bubnikv