prusa3d/PrusaSlicer #1312: Print quality regression above supports

stahlfabrik, 2018-10-11:
### Version
1.41.1-rc

### Operating system type + version
Mac OS High Sierra

### Behavior
I had high hopes for the new version of slicer cause it supposedly improves print quality above support material. I compared with Neferiti's chin. I printed her fully with the last stable version. I am not a fan of how the support looks with the old version - but is very visibly better than the quality that the new slicer version produces.

I think it is a recession. Especially as I expected it to improve, based on the release notes.

#### STL/Config (.ZIP) where problem occurs
[bust-of-nefertiti-at-the-neues-museum-berlin-1_0.15mm_PLA_MK3.gcode.txt.zip](https://github.com/prusa3d/Slic3r/files/2470535/bust-of-nefertiti-at-the-neues-museum-berlin-1_0.15mm_PLA_MK3.gcode.txt.zip)
[image]

stahlfabrik, 2019-01-11:
Although the quality in that case was less nice I understand that it might be cause the algorithm now works better?!:-) Well I played with support settings to achieve better results. So there is no issue - but maybe the distance in the default profiles should be reduced a hair?

bubnikv, 2020-11-30:
I don't think there was a change in the support logic. Anyways, the issue is very old.

closed 2020-11-30 by bubnikv