prusa3d/PrusaSlicer #1313: Print Speed set to Retraction Speed??

AbeFM, 2018-10-11:
### Version
1.41.1b (MMU)

### Operating system type + version
win10x64

### Behavior
Printing speed sets itself from 100% to Extruder Retraction Speed. Each time a filament change finishes, the printing speed sets itself to 35.
I changed the retraction speeds to (Ext1,2,3...) 36,37,38,39,33 respectively.
I came in and saw printer at 35%, overrode it to 100.
F2: 36%
F3: 37%
F4: 

So it seems it's grabbing the Retraction Speed from each Ext(n-1) and setting that as the speed when it is loading the filament.

I think that's a bug.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[SpeedsChanging.zip](https://github.com/prusa3d/Slic3r/files/2471084/SpeedsChanging.zip)

AbeFM, 2018-10-23:
I keep seeing this, and variations on it. I've tried posting in another area, and I don't know what I should do.

My prints take >3x as long as quoted. I have more observations which could help, but I'm yelling into the wind here.

rtyr, 2020-05-24:
35% speed was used for flexible filaments by design, it was not related to retraction speed. Anyway, this behavior is completely reworked now so this issue is no longer valid. Closing.

closed 2020-05-24 by rtyr