prusa3d/PrusaSlicer #1317: Why does M73 truncate the minutes?

eyal0, 2018-10-12:
### Version
https://github.com/prusa3d/Slic3r/commit/4009fdcc1850ce8db5090f401db40a079fe81946

### Operating system type + version
Linux

### Behavior
M73 progress reports are truncated to whole minutes.

Why?  I would expect the output to include fractions for better accuracy.

eyal0, 2018-10-12:
Thanks for your quick reply.

I see the code in the firmware.

The code first [calls code_value which returns a float](https://github.com/prusa3d/Prusa-Firmware/blob/9c1ef5e414e1e7754a6fe65fb1ed25ba6d3f251a/Firmware/cmdqueue.h#L76).  That float is then [stored in `print_time_remaining_normal`](https://github.com/prusa3d/Prusa-Firmware/blob/bbd4f70f41bfcb69be0494bca8a6052e429632f8/Firmware/Marlin_main.cpp#L5352), which is a [uint16](https://github.com/prusa3d/Prusa-Firmware/blob/bbd4f70f41bfcb69be0494bca8a6052e429632f8/Firmware/Marlin_main.cpp#L479).  And then it is display [here](https://github.com/prusa3d/Prusa-Firmware/blob/bbd4f70f41bfcb69be0494bca8a6052e429632f8/Firmware/Marlin_main.cpp#L5358).

Because the prusa printer is already converting double to uint16, it should not matter if you send a double or an int, Prusa firmware will convert it to int and display it as int.  So for this reason, I think that you could add decimal places to the M73 output without any impact on the display.

The gcode would grow in length, that is true.  If you added two decimal places for each M73 line, then the gcode would grow by 3 characters per M73 line and be accurate to within a second.  On these [these gcode files](https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/79#issuecomment-428950510), the growth would be just 0.01% of file size, which is very small.

So I think that it might still be a good idea.  Especially because the truncation is causing the M73 line to be wrong by up to 60 seconds each time.

eyal0, 2018-12-18:
First, I think that there is a bug.  The M73 line is printed once per minute but it is not printed when the seconds are 0.  So it prints M73 at, for example, 10min43sec, then 9min43sec, then 8:43, etc.  So the accuracy is off though it's only by a consistent amount.

[PTG](https://github.com/eyal0/OctoPrint-PrintTimeGenius) is post-processing those inserted M73 lines to provide a more accurate time estimate for OctoPrint's display.  If the seconds were available, it would be even best.

It would also be nice to be able to specify the interval.  I guess that it wouldn't be useful for M73 but potentially Slic3r could insert comments into the code after every X seconds of printing saying how much time has elapsed.  This is what Cura does, though it does it per-layer, not per X seconds.

rtyr, 2021-01-21:
I believe it was answered. Current M73 format is currently also supported by Marlin FW.

Closing.

closed 2021-01-21 by rtyr