prusa3d/PrusaSlicer #1322: MMU2 Printer Rrofile does not show user defined Filament profiles

Petclaud, 2018-10-14:
### Version
1.41.1 RC + win64

### Operating system type + version
W10

### Behavior
I have an MK2.5 and an MK3 with MMU2 and use Slic3r for both printers and for single and multimaterial prints. In the meantime I have made a quite a number of profiles for the various filaments I have.

When I select MMU2 under "Printer Settings", I cannot select any of my own filament profiles - only Prusa provided profiles (Verbatin, Prusament, Generic, etc.). When I enable "Show incompatible print and filament presets" under "Preferences" than also my own profiles do not show up.
Selecting MMU - single as printer setting, all my custom profiles do show up.

I wonder why this is programmed like this and would opt that user made filament profiles should always be selectable. With the current set-up one needs to again make user defined profiles when printing with the MMU printer profile which is very annoying.

In addition: if the user selects filament profiles that are incompatible with each other (for example an ABS profile and a PLA profile - which will not work because of the different temperatures) that the user is notified that the temperature range is too high. I would also opt for a 10 degrees or so relaxation for this for the extruder temperature since the extruder is able to quickly heat and cool down and for sure during the wiping the extruder will reach its target temperature. 

Is this a new feature request? YES

MK3 Single Printer Profile:
[image]

MMU2 Printer Profile:
[image]

Petclaud, 2018-10-16:
Thx Bubnikv for the fast reply. So if I understand correctly, we should always keep two versions of the filament profiles; one with the default printer condition:

"printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and single_extruder_multi_material"

and one with the MMU2 compatible condition:

"printer_notes=~/.*PRINTER_MODEL_MK3.*/ and single_extruder_multi_material"

**OR**, wil my MK2,5 printer also accept the the last printer condition (the MMU2 compatible).

So, I really want to avoid replicating profiles; cooling fan settings, cooling thresholds and extrusion multipliers can vary filament to filament and hence keeping just one filament profile per filament that can be used for both  MMU2 and none-MMU2 prints is favorable to me :-)

I understand that the ramming settings differ from MMU1 to MMU1, but here I am talking about one MK3 printer with an MMU2 and a MK2.5 without MMU2 - and I guess that there are various other users with more than one MK2/2.5 and/or MK3 which have only one printer equipped  with the MMU2 and hence probably also just want one filament profile for all their printers.

I have only limited experience with the MMU2 yet, but what I did experience is that I had 4 consecutive fails printing the same model using 3 different colors of (near-new) Filamentum PLA spools and all failed due to clogging in various places and stringing (very loooong +/- 10 cm) fluffy strings in the tube between the extruder and the MMU2 and only once I raised the extruder temperature  on the 5th attempt to 220 degrees (yes I know it is a tad high for Filamentum), the print actually finished and came out great. Maybe this was just luck, 

Reason for the suggestion of allowing various extruder temperatures is the optimum temperature for filaments vary brand by brand and of course also for different materials, I assume that users want to be able to set the temperature individually for each of the 5 filament channels. Filamentum's PLA has a print temperature range of  ± 190 - 210° C and Formfutura for example specifies a Print temp: ± 180 - 220° C for their PLA. Maybe Filametum performs better at 220 and Filamentum at 205.... Also for mixing materials this might be needed, eg PLA with TPE, or another flexible material and/or soluble materials. 

I hope this clarifies.

bubnikv, 2018-10-17:
*> OR*, wil my MK2,5 printer also accept the the last printer condition
(the MMU2 compatible).

We are currently updating our Slic3r profiles for MK2.5 MMU2.

> So, I really want to avoid replicating profiles; cooling fan settings,
cooling thresholds and extrusion multipliers can vary filament to filament
and hence keeping just one filament profile per filament that can be used
for both MMU2 and none-MMU2 prints is favorable to me :-)

We avoid the duplication by inheritance in the single profile file. We
would like to make it smarter (easier to adapt from the user interface),
but it will take some development time to get there.

Petclaud, 2018-10-17:
`We avoid the duplication by inheritance in the single profile file. We
would like to make it smarter (easier to adapt from the user interface),
but it will take some development time to get there.`

Great, looking forward to it

AbeFM, 2018-10-18:
Would it be possible/useful/worthwhile to have checkboxes for the already-installed-printers under filament, so we could check/uncheck easily to make material compatible more easily?

bubnikv, 2018-10-18:
It is there at the Filament -> dependencies page at the top, it works for
the compatible printers, but it will most likely go away soon, as the
condition is easier to work with when handling a lot of presets.

AbeFM, 2018-10-18:
Thanks! I didn't realize the "all" and the textbox beneath were exclusive.

What I don't like is that I have it checked for "all printers" but the filament dissapears - making me think the box may have been populated (with "All" the existing printers) *before* I added the new printers, so they aren't included in "all".

If I check "ALL", that filament should be always available to me, and it appears not to be.

bubnikv, 2018-12-12:
I believe this issue was mostly about clarifying how the "compatible printers" and the "compatible printer condition" works, and why we split the MMU2 filament profiles from the non MMU2 filaments. Closing.

closed 2018-12-12 by bubnikv