prusa3d/PrusaSlicer #1323: GUI widgets not drawing on Arch Linux

stuartpb, 2018-10-14:
### Version

1.41.1-rc

### Operating system type + version

Arch Linux

### Behavior

[image]

Note how everything except icons, text, and the plater itself is completely missing - no button borders, no drop-down buttons, no checkboxes, nothing.

Here are the errors I get in console:

```
/tmp/.mount_Slic3rC132Lw/usr/bin/slic3r.wrapper: line 125: 32119 Aborted                 (core dumped) LD_LIBRARY_PATH="" zenity --question --title="$TITLE" --text="$TEXT" 2> /dev/null
/tmp/.mount_Slic3rC132Lw/usr/bin/slic3r.wrapper: line 125: 32124 Aborted                 (core dumped) LD_LIBRARY_PATH="" zenity --question --title="$TITLE" --text="$TEXT" 2> /dev/null
Creating main frame...

(slic3r.pl:32083): Gtk-CRITICAL **: 11:28:34.612: IA__gtk_color_button_set_color: assertion 'color != NULL' failed

** (slic3r.pl:32083): WARNING **: 11:28:34.869: Pixbuf theme: Cannot load pixmap file /usr/share/themes/Adwaita/gtk-2.0/assets/treeview-separator-ltr.png: Couldn’t recognize the image file format for file “/usr/share/themes/Adwaita/gtk-2.0/assets/treeview-separator-ltr.png”

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

** (slic3r.pl:32083): WARNING **: 11:28:34.869: Invalid borders specified for theme pixmap:
        /usr/share/themes/Adwaita/gtk-2.0/assets/treeview-separator-ltr.png,
borders don't fit within the image

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:34.869: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
[... 2830 lines cut ...]

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_n_channels: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_pixels_with_length: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_rowstride: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_width: assertion 'GDK_IS_PIXBUF (pixbuf)' failed

(slic3r.pl:32083): GdkPixbuf-CRITICAL **: 11:28:35.006: gdk_pixbuf_get_height: assertion 'GDK_IS_PIXBUF (pixbuf)' failed
```

stuartpb, 2018-10-14:
[This comment](https://github.com/jeromerobert/pdfshuffler/issues/3#issuecomment-347505014) also suggests the issue may stem from the necessary loader not being statically included.

yschroeder, 2018-10-16:
I played around with the themes in Gnome Tweak Tool and it did not change anything.

Also I looked into changing the GTK+ 2 theme for Slicer directly:
https://wiki.archlinux.org/index.php/GTK%2B#Themes
But that also did not work.

stuartpb, 2018-10-16:
@stelgenhof Have you run `pacman -Syu` recently? What are your GTK / libpng versions?

stelgenhof, 2018-10-16:
@stuartpb I actually compiling/running from source. Yes, updating the OS/packages everyday :)

This is what is installed on my machine
```
- extra/libpng 1.6.35-1
- extra/gtk-engines 2.21.0-3
- extra/gtk2 2.24.32-1
```
Using the Arc Theme: `arc-gtk-theme 20180715-1`

stelgenhof, 2018-10-18:
I am running on a Thinkpad T410 (Intel Core i5 M520 2.4GHz, 8GB RAM, Graphics: Intel Ironlake Mobile)

vojtechkral, 2018-10-18:
I am running Arch as well, not seeing the problem on my end. GTK2 theme is Adwaita (both regular and dark variants seem to work). Intel GPU.

```
$ pacman -Q libpng gtk-engines gtk2
libpng 1.6.35-1
gtk-engines 2.21.0-3
gtk2 2.24.32-1
```

vojtechkral, 2018-10-18:
`libpng` is not built-in / linked statically either, but I am noticing that the wxgtk lib dynamically links two versions of `libpng`, `libpng12.so.0` and `libpng16.so.0`, which is a bit suspect.

ccavanaugh, 2018-10-26:
If it helps, building slic3r-prusa3d from AUR fixes the issue.

thilo-adk, 2018-12-03:
Hi,
with Manjaro Cinnamon i have the same problem. With XFCE / Gnome / Mate it works fine.

bubnikv, 2019-05-21:
Does the PrusaSlicer-2.0.0 work for you? It shall not contain the libpng, so let's hope for the best.

stelgenhof, 2019-05-21:
@bubnikv Let me compile v2.0 on my machine and let you know if it has been fixed.

stuartpb, 2019-05-21:
FWIW, I used the AppImage for one of the betas with the new UI before the rename to PrusaSlicer 2.0, and rendering there worked fine, so I believe this has effectively been fixed.

stelgenhof, 2019-05-21:
Although compiling from the source code caused some errors, I can confirm that using the AppImage the rendering works fine on Arch Linux.

lukasmatena, 2019-05-22:
@stelgenhof Let's discuss the error in https://github.com/prusa3d/PrusaSlicer/issues/846, where you mentioned in as well.

Since the issue has apparently been fixed (as confirmed by @stuartpb ), I'm closing the issue. If anyone disagrees, it will be reopened.

closed 2019-05-22 by lukasmatena