prusa3d/PrusaSlicer #1324: Support Enforcers Driven By Overhang Threshhold Value

chrisprad, 2018-10-15:
### Version
1.41.1-rc+win64

### Operating system type + version
Windows 7 Home Basic 64-bit

### Behavior
slic3r seems to use the Overhang threshold value, even when no longer generating supports and using a support enforcer. 

My support enforcer:
[image]

Settings at 15% overhang:
[image]

Result:
[image]

Settings at 80% overhang:
[image]

Result:
[image]

I'm not sure if the support enforcer following the overhang threshold is intentional or not. If it is, the field shouldn't be disabled to modification in the settings menu. If it isn't, then the value should always generate supports (which is how I read the description of support enforcers).

#### STL/Config (.ZIP) where problem occurs

[star-wars-darth-vader-report.zip](https://github.com/prusa3d/Slic3r/files/2477023/star-wars-darth-vader-report.zip)

Settings named "0.10mm DETAIL MK3 report" in attached zip.

bubnikv, 2018-10-15:
Thanks for the report. You are right that the support enforcers should not be sensitive to the overhang threshold value. We will most likely not fix it in the 1.41.1 release.

bubnikv, 2021-09-23:
I believe this issue has been fixed long time ago.

bubnikv, 2021-09-23:
unfortunately not. The issue here most likely is, that the overhang islands are very tiny thus they are removed. The filtering happens even for paint on supports.

bubnikv, 2021-09-23:
It is similar to https://github.com/prusa3d/PrusaSlicer/issues/1190