prusa3d/PrusaSlicer #1325: Compile Error

stelgenhof, 2018-10-15:
### Version
`git describe --tag`  version_1.41.1-rc-3-gc06cf009e

### Operating system type + version
Arch Linux 4.18.8-arch1-1-ARCH

### Behavior
Can't load '/home/sacha/Code/mySlic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /home/sacha/Code/mySlic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so: undefined symbol: _ZN5boost6system15system_categoryEv at /usr/share/perl5/core_perl/XSLoader.pm line 93.
 at /home/sacha/Code/mySlic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/Slic3r/XS.pm line 24.
Compilation failed in require at /home/sacha/Code/mySlic3r/lib/Slic3r.pm line 50.
BEGIN failed--compilation aborted at /home/sacha/Code/mySlic3r/lib/Slic3r.pm line 50.
Compilation failed in require at slic3r.pl line 16.
BEGIN failed--compilation aborted at slic3r.pl line 16.
Segmentation fault (core dumped)

bubnikv, 2018-10-15:
Most likely you are mixing different versions of boost.

mloidl, 2018-10-15:
@stelgenhof Please clean your build env and run cmake again. I had something similar on arch.

stelgenhof, 2018-10-15:
@mloidl Ah I remember know as I had this before. Let me clean my build env again.