prusa3d/PrusaSlicer #1326: Bug: Support enforcer has zero impact

apphead, 2018-10-15:
### Version
1.41.1 rc win64 (commit b8c2566c447b3c261a092ce9409c791048114ffb)

### Operating system type + version
Win8.1 x64 (updated just now)

### Behavior
* support enforcer does not enforce support (while the support blocker seem to work)
* auto generated supports on or off does not make a difference, neither does the support everywhere or on build plate change the behaviour of the enforcers...

[image]

#### STL/Config (.ZIP) where problem occurs
[Config and stl.zip](https://github.com/prusa3d/Slic3r/files/2478168/Config.and.stl.zip)

chrisprad, 2018-10-16:
Per my own experience in https://github.com/prusa3d/Slic3r/issues/1324,
I'm guessing slic3r applies a lot of the settings in the supports tab within the support enforcer. Try changing both the support to everywhere and the overhang threshhold to 80 or 90 without automatic supports off. 

Maybe that can help you do a temporary workaround.

bubnikv, 2018-10-16:
Yes, this issue is similar to #1324.

The issue is a bit more complex though. For example, in #1324 the stick
should be supported automatically. There is a logic in action though, which
filters tiny supports in each layer to reduce "support noise", but this
logic is local to a layer, therefore not being aware of the stick structure
spanning multiple layers. I would rather keep this issue open for now and
fix it together with the improvements of the small island filtering.

GWdd, 2019-06-17:
Related to this is that Support Enforcer also seems to use Don't Support Bridges as a filter.   It would seem logical that if a user creates an enforcer, the user wants that area to be supported, regardless of other settings.

I tried multiple variations and the only factor was Don't support bridges.

[image]
[image]
[image]

bubnikv, 2021-09-23:
> Related to this is that Support Enforcer also seems to use Don't Support Bridges as a filter. It would seem logical that if a user creates an enforcer, the user wants that area to be supported, regardless of other settings.

Implemented in PrusaSlicer 2.4.0-alpha1.

closed 2021-09-23 by bubnikv