prusa3d/PrusaSlicer #1327: Weird behavior with 0.20mm layer height

Saij, 2018-10-15:
### Version
1.41.1-rc+win64

### Operating system type + version
Windows 10 64-Bit (newest Update)

### Behavior
As soon as I try to slice the STL file you find down below (MOD_Armrear01) with a layer height of 0.20mm I get some weird artefacts on the else flat surface.
[image]

But if I slice it with 0.15mm or 0.10mm layer height everything is perfectly fine.
[image]

First I thought that the model might be defective. But after slicing it with 0.10mm layer height it was correct. Weird behaviour I think.
What can I do to fix this?

#### STL/Config (.ZIP) where problem occurs
[Config 200µm](https://github.com/prusa3d/Slic3r/files/2479329/Config200.zip)
[Config 150µm](https://github.com/prusa3d/Slic3r/files/2479330/Config150.zip)
[Config 100µm](https://github.com/prusa3d/Slic3r/files/2479331/Config100.zip)
[MOD_Armrear01](https://github.com/prusa3d/Slic3r/files/2479332/MOD_Armrear01.zip)

Saij, 2018-10-16:
What's weirder is that with a layer height of 0.201mm it is fine. So I think there might be a rounding error of some kind?
[image]
[image]

bubnikv, 2018-10-16:
This is quite an evergreen, it should be described in a FAQ.

This usually happens if the top layer height is very much at the height of
the last slicing plane. Ideally you should avoid that when designing your
models.

For example, a cube at 10.1 mm height may show such issues if sliced at
layer height of 0.2mm, as the last slice will be calculated exactly at 10.1
mm. Due to numerical errors, part of the slice may be above and some may
become below the cutting plane.

You will most likely fix your issue, if you increase or decrease your layer
height just a bit, then the last slicing plane will not match the object's
top plane and everything will be golden.

Saij, 2018-10-16:
Thank you very much for this answer. Will try and print it with 0.1999mm layer height as that gave the best result in Slic3r preview.

closed 2018-10-16 by Saij