prusa3d/PrusaSlicer #1328: Overextrusion on infills and solids

GizmoTheGreen, 2018-10-15:
### Version
1.4.1+win64

### Operating system type + version
Windows 10 Pro

### Behavior
Experiencing over extrusion on infill/solid layers, while perimeters are perfect.
Extrusion ratio set to 1, nozzle set to 0.4, and material thickness to 1.75.

The issue is if I change any of those to decrease extrusion amount (either increase materia thickness or decrease extrusion ratio) the problem reverses, infill is alright but the walls become too thin and don't even touch each other. menaing I can peel one layer of wall off leaving the other, then repeating.

Will try to upload example images.
[image]
[image]
[image]

please observe I have double checked my e-steps for x,y,z and extruder. a 20 and 40mm testprint measures correctly.

apparantly this was an issue in regular slic3r at one point? https://github.com/slic3r/Slic3r/issues/2960

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/prusa3d/Slic3r/files/2479857/config.zip)

bubnikv, 2018-10-16:
There is a little bit of an intentional overfill due to the perimeter / infill overlap to compensate for the tiny gaps where the infill lines fuse with the inner perimeter. You may try to decrease that.

AbeFM, 2018-10-18:
Gizmo: Is this new? Since I got the MMU (and flashed the latest release firmware) I've noticed a bit of the same. I felt a little unsure bringing it up as I am using the MMU and its myriad of issues leaves me unclear what's a cause and what's an effect.

supermerill, 2018-10-24:
note: I have created a "exact fill density" feature in my fork. It reduce/increase the flow/thickness, comparing the volume of extruded plastic and the volume of the space to fill.

It's a bit extensive as a modification, and not extra-useful, so i didn't bothered you with it. But if you want, i can create a pr.

bubnikv, 2018-10-24:
Thanks for your effort. As usual, we are busy with refactoring the rest of
the application into C++, with the SLA supports, with the UI improvements.
We will happily let our testers to evaluate your prototype, but it is too
early for a PR. My rough (unfortunate) estimate is that nobody will have
time to review this PR until the beginning of the next year.

bubnikv, 2020-03-20:
I reviewed the code by @supermerill and I don't find it a good idea. @supermerill calculates the amount of material required to fill in a region, then he sums up the amount of material to be extruded into infill, and then he increases extrusion amount of all the infill lines proportionally to reach the target extrusion amount of the region to be filled in. While this may sound a good idea and it would certainly work if we were pouring water, I think this approach is wrong in general for FDM infills.

I am not sure how to help @GizmoTheGreen, I would tend to close this issue.

jflyer81, 2020-04-16:
Would it be possible to add a setting to adjust infill and perimeter extrusion multipliers independently? Similar to what is discussed [here](https://github.com/slic3r/Slic3r/issues/2441), [here](https://github.com/slic3r/Slic3r/issues/2443), and [here](https://github.com/slic3r/Slic3r/issues/3100). 
Alternatively, in the Slic3r manual there's [a line spacing section](https://manual.slic3r.org/advanced/flow-math) that discusses the side-by-side line overlap, that uses an overlap_factor from 0 to 1, with 1 as the default. An option for the user to adjust this factor for solid infill could have the desired effect.

rtyr, 2020-10-20:
I believe we can close this issue. Decreasing infill/perimeter overlap and using linear advance should help (assuming that extrusion multiplier is OK).

closed 2020-10-20 by rtyr