prusa3d/PrusaSlicer #1329: i am having y stepping errors  i think that it is slc3r related

makestuf, 2018-10-16:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-10-16:
Sorry, but the only information you provided is 
> i am having y stepping errors i think that it is slc3r related

which is a little too little to draw any conclusion. First, I don't think you are right, but if you do believe there is an issue in Slic3r, you have to provide more information, otherwise we should close this issue.

bubnikv, 2018-12-12:
No response, closing.

closed 2018-12-12 by bubnikv