prusa3d/PrusaSlicer #1331: Slic3r should check users model is merged with purge block

mylife4aiurr, 2018-10-17:
Bug:
Slic3r should check if purge block is merged with users model in mmu2 prints, and not allow this. Currently slic3r will merge skirt and model into purge block on mmu2 prints and not warn user, so you get this:
[image]
[image]

This model took more than a day of printing time
@Kuntry3d

1.41.1-rc+win64
Win10

AbeFM, 2018-10-18:
Similarly, the skirt can run under the block - and supports also.

All three should be looked for. The skirt, in particular, can just be set up to include the block.

lukasmatena, 2019-08-26:
> the normal pre-slicing checks (and skirt planning) done

The "normal" checks do not include (and never included) intersection of objects. This issue is not specific to the wipe tower. Closing as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2501 and https://github.com/prusa3d/PrusaSlicer/issues/2378

closed 2019-08-26 by lukasmatena

lukasmatena, 2019-08-27:
> One could argue that the block is a special case

In that case supports are also a special case (they also do not show in 3D view, not even after slicing). Which makes the case not so special.

> slicer should check 4 and either correct or at least warn about.

Possibly it should, I'm not arguing about that. We may improve this in future. I just don't think we need another issue to track it, having https://github.com/prusa3d/PrusaSlicer/issues/316, https://github.com/prusa3d/PrusaSlicer/issues/1275, https://github.com/prusa3d/PrusaSlicer/issues/684 and https://github.com/prusa3d/PrusaSlicer/issues/2598 all open.