prusa3d/PrusaSlicer #1332: Set bed temperature in printer profile, at least for MMU prints

AbeFM, 2018-10-18:
### Version
Current

### Operating system type + version
NA

### Behavior
If you slice an MMU print, and have materials with different bed temperatures, the bed temp swings up and down with each one.

I think something like a setting "use filament settings", "use highest" or "use average" temperature would be better than ramping up and down. An exception may be on the first layer, I'm unsure.

Motivation: I'm using PLA as breakaway support for PETg and the bed swings +/- 30 degrees once per layer... So I end up with materials like "PETg 235noz 70bed", and by the time I'm done with all the permutations I have a ridiculous list I can't find anything on.

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs
NA

bubnikv, 2018-10-18:
> Motivation: I'm using PLA as breakaway support for PETg

How well does it work?

AbeFM, 2018-10-18:
It works great. Peels right off. I was having issues with weak parts and
under extrusion I don't understand but you get nice surfaces. Obviously you
still have to be able to reach it, but thin horizontal shelves come out the
right side instead of an extra (half) layer thick.

That said, the bed issue is why I'm writing. Post on the forums for
printing tips, I'm there too.
                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

bubnikv, 2018-10-18:
Would you please post your profiles? How do you set the supports? Similar
to the same material support, or are you using the soluble supports
settings?

ruedli, 2018-10-18:
I am not sure whether this is a good feature:
- what if you _want_  different temperatures per filament?

As an alternative - when you want the same temperature for all materials used- : why not create material profiles with the same temperature and select those when slicing?

AbeFM, 2018-10-18:
Please pm me or something. This isn't a thread about support material.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2018-10-19:
> I am not sure whether this is a good feature:
Reasonable to disagree. Please explain WHY you would want to rock the temperature back and forth many times per layer as you print. Does tweaking the bed temp when you're at layer 100 mater?

 
> * what if you _want_  different temperatures per filament?
I mentioned: If you slice an MMU print, and have materials with different bed temperatures, the bed temp swings up and down with each one.
Also, you'll note that I *specifically* menioned that you could : something like a setting "use filament settings", "use highest" or "use average" temperature 

> As an alternative - when you want the same temperature for all materials used- : why not create material profiles with the same temperature and select those when slicing?

 I end up with materials like "PETg 235noz 70bed", and by the time I'm done with all the permutations I have a ridiculous list

You literally asked me about every line of my first post. I'm unclear what you're asking, as I see it I already discussed all of those. I'm more than happy to discuss but I do not understand the question.

In case you don't know why I said it: If you're printing with something like PET-G, and you drop the temp to <60 degrees, the part could come off the bed.
If your bottom layer is mostly PLA, PVA, then ramping up to tripple digit temps could liquify and unstick the part.

ruedli, 2018-10-20:
I agree with AbeFM, when managing the bed temperature, the correct setting for ensuring bed adhesion depends on the material in the first layer and not not on "some" material used down the road.

If it is selected from the material parameters, it should take into account first layer(s) only, -and/or- let the user decide by overriding these settings.

bubnikv, 2018-10-22:
Thanks for your hints. I tend to agree with you, though I am not sure how top priority this issue is, as our profiles for both PVA and BVOH set the bed temp to 60 and nozzle temp to 210+-5.

AbeFM, 2018-10-22:
If the printer only works with your filaments, that's a "closed ecosystem". I do not feel it is consistent with the open source nature of the rest of the project.

bubnikv, 2018-10-22:
@AbeFM 
> I do not feel it is consistent with the open source nature of the rest of the project.

With all respect, you seem to not get the "open source" concept. Open source is about your access to the source code, and you having the option to modify the source code. We provide what we have tested, and we also provide all the source code and design files. You are free and most welcome to change anything you need.

AbeFM, 2018-10-23:
You're right, that was not the response you deserved.

The right response is:
The problem is real, just because you personally run across it personally doesn't invalidate my suggestion.

It is possible too print with materials which have different default bed temperatures.

There is no reason (that I've seen suggested yet) to ramp the bed temperature cyclically.

A reasonable method would be as I, and others, have suggested.

Can we spend more time fixing the software and less bickering over phraseology? Please?

AbeFM, 2019-08-01:
Bump. I still have to create 3 or 4 custom filaments for every print I do that involves multiple materials.

I've seen this requested over and over on forums and here.

rednaxal, 2023-02-10:
Having the same issue. It makes no sense to me to change the bed temperature upon tool change. The bed temperature must always be set for the first layer material and never altered later during a print. 

In the meantime I think the temporary solution is to use the g-code substitutions in the print settings -> output options. 
Find (mark as RegExp): `  M140 S\d* ; set bed temperature`
Replace with: ` ;`

P.S. Ensure that in your start gcode the bed temperature is not set with exactly the same line.

Hausmeister486, 2024-06-23:
Why not use a central "Object Temperature Override" in the print settings for the print bed temperature? Then the print bed temperature is overridden for this object regardless of the separate filament print bed temperature settings.

Samuel1698, 2024-08-30:
Bumping this issue. Had 2 prints fail because the PETG support interface I had would increase the bed temp too much and the PLA slid off the bed mid print.

closed 2026-09-29 by prusaslicer-bot[bot]