prusa3d/PrusaSlicer #1334: Lost all system presets and many user presets of "Print settings" from one start to another

Idus, 2018-10-18:
### Version
1.41.0+win64

### Behavior
* THE PROBLEM
From one time to the other - all system presets and many user presets of "Print Settings" profiles was not loading any longer - but all user presents (10 in total) are still in the AppData folder (...\AppData\Roaming\Slic3rPE\print).

* WHEN?
Out of the blue - I have not managed Print settings profiles for severel prints. But this time when I opened a STL - they were just not there any longer.

[image]

bubnikv, 2018-10-18:
Would you please zip the content of your Slic3rPE directory and attach it
here? Please make sure there are no OctoPrint profiles stored there.

You will find the Slic3rPE directory under
C:\Users\YOUR_USER_NAME\AppData\Roaming\Slic3rPE

Thanks

Idus, 2018-10-18:
Here it comes... will you know my bank ID and girlfriends maiden name now as well? ;-)

Took it as-is - no changes...

[Slic3rPE.zip](https://github.com/prusa3d/Slic3r/files/2491943/Slic3rPE.zip)

FidelCapo, 2018-12-13:
Thank you. I have found that if the printer preset "IDUS Prusa MK3 + OctoPrint Idus v2" is selected, the system print presets are not displayed. The "IDUS Prusa MK3 + OctoPrint Idus v2" printer profile has a modified nozzle diameter and therefore does not match the system presets compatibility conditions. When I set the nozzle diameter back to 0.40, the system presets are displayed. Can you confirm it? Thank you.

Idus, 2019-02-02:
We´ve solved this in another thread - it was because of the conditioning in the PRINT preset that voided any deviations in the PRINTER preset. Thank you!

closed 2019-02-02 by Idus