prusa3d/PrusaSlicer #1335: Printer Profiles affecting Print Settings Parameters

sarf2k4, 2018-10-18:
### Version
1.41.0

### Operating system type + version
win 10 64 bit

### Behavior
I'm unable to change any parameters in "print settings" for custom "printer settings". Both are inherited from "default" preset, even inherited from mk3 print setting won't do anything especially infills.

I tried one of the other printer setting "generic" profile, it is able to change the parameters. Both printer profiles inherited from default profile except "generic" were created before 1.41.0 while "tantillus" were created on this version of slic3r

#### STL/Config (.ZIP) where problem occurs
[tantillus.zip](https://github.com/prusa3d/Slic3r/files/2490542/tantillus.zip)

bubnikv, 2018-10-18:
I don't quite understand your request, but anyway:

Your 0.2mm Tantillus.ini contains the following fields of interest:
compatible_printers = Tantillus
compatible_printers_condition = printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/
and printer_notes=~/.*PRINTER_MODEL_MK3.*/ and nozzle_diameter[0]==0.4
inherits = 0.20mm FAST MK3

that means, it is inherited from 0.20mm FAST MK3 profile and it is set to
be compatible with the "Tantillus" printer. You may need to clear the
"compatible_printers_condition" for the "compatible_printers" to work.

sarf2k4, 2018-10-18:
The main problem here actually is that, the printer settings "tantillus" locks the print settings to "default" and that I can't change any parameters effectively, while "generic" didn't lock the print settings. 

Can you check what's happening?

I already have second "tantillus" derived from "generic" profile from printer setting

As for the compatible printers parameter under "dependencies". I also tried to clear the condition, it is still the same, locked to default

FidelCapo, 2018-10-18:
Can you please zip the content of your Slic3rPE directory and attach it
here?

You will find the Slic3rPE directory under
C:\Users\YOUR_USER_NAME\AppData\Roaming\Slic3rPE

Thank you

sarf2k4, 2018-10-18:
[Slic3rPE.zip](https://github.com/prusa3d/Slic3r/files/2490727/Slic3rPE.zip)
Here you go

rtyr, 2020-05-24:
I believe this old issue is no longer valid. I tried it with current 2.2.0 and I was able to edit values. Closing.

closed 2020-05-24 by rtyr