prusa3d/PrusaSlicer #1338: Sequential Printing Priority menu item is missing

marclefevre, 2018-10-18:
### Version
1.41.0+win64

### Operating system type + version
Win64 Pro
10.0.17134 Build 17134

### Behavior
Place several objects on the plater either by replication or multiple ADD operations.
Select "Complete Individual Objects" in the Print Settings/Output options menu
In the Plater/3D view, right click on any item and select "Settings"
Click on the green circle/plus icon and select "Advanced"
The user manual suggests that this menu should contain the Sequential Printing Priority option but it does not.

* _Expected Results_
Expected menu item list to contain "Sequential Printing Priority"
* _Actual Results_
Menu only contains "Bridge flow ratio", "Elephant foot compensation", "Infill/perimeters overlap", and "XY size compensation"
  * _Screenshots from __*Slic3r*__ preview are preferred_

Manual:
[image]

What I see:
[image]

_Is this a new feature request?_
No, unless the manual describes an unreleased feature.
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
Any of my STL files show this behavior.

marclefevre, 2018-10-18:
I made a mistake in the screenshot for the manual.  Here is the correct one:
[image]

chrschn, 2019-03-05:
Sequential printing priority was added to upstream Slic3r back in 2017:
https://github.com/slic3r/Slic3r/pull/4153/files/29fac5c1fe1acad4f0b28cb222524c9bc9726b4f

It seems that it was never merged into Slic3r PE, that's why the option is missing.

bubnikv, 2019-08-01:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/486

closed 2019-08-01 by bubnikv