prusa3d/PrusaSlicer #1340: supports on build plate only bug

scheric, 2018-10-18:
### Version
1.41.0 x64

### Operating system type + version
windows 10 home x64

### Behavior
* there is a support structure on the object while the supports have been set to bed only
* to reproduce: enable supports and enable sheath around supports
[image]
* I excpect that there is no support on the object 

* _Actual Results_
[image]

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
[Nieuwe map.zip](https://github.com/prusa3d/Slic3r/files/2493204/Nieuwe.map.zip)

scheric, 2018-10-19:
I tested some more settings and the results are even more wrong then before.

If you disable sheath around infill and set pattern spacing to 4 and the rest the same as picture.
This will happen:

[image]
[image]

I have tested this on the portable version of slic3r PE version 1.41.1-rc.
The same results.
Am i doing something wrong?

FidelCapo, 2018-10-22:
Thank you for report. This is because of how the algorithm for calculating support works. I thing the algorithm is trying to stabilize the support and depends on pattern spacing too because slicer do not cut support in middle of space between supports. That is why they can sometimes appear where they are not expected.

mdealer, 2018-12-07:
This always appears on the object for me no matter what I change. I use 1.41.1.

[image]

[image]

scheric, 2019-06-09:
# update with PrusaSlicer 2.0.0

### The problem is still there. 
[image]

### support settings
[image]

timur-tabi, 2019-07-24:
I have just encountered this problem with v2.0.0+.  Steps:

1. Download https://www.thingiverse.com/download:4890219
2. Set to "Generic PLA and 0.2mm QUALITY" print settings.
3. Scale to 103.5%.
4. Slicer with supports set to "build plate only".

[image]

scheric, 2019-07-25:
### This issue has a duplicate
 #915

GarthSnyder, 2020-01-20:
Still there in 2.2.0-alpha2+. See attached project.

[image]
[BuildPlateOnly.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4084589/BuildPlateOnly.3mf.zip)

theDoktorJ, 2020-04-18:
This is still happening in 2.2 full release.  
[Hero_Me_Dual_5015_Duct_Gen3.zip](https://github.com/prusa3d/PrusaSlicer/files/4496486/Hero_Me_Dual_5015_Duct_Gen3.zip)
[image]
[image]

bubnikv, 2021-03-18:
Tested with the current master.

The support columns are trimmed to not land on top of an object. However this may make the column too small to be stable.

[image]

[image]

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv