prusa3d/PrusaSlicer #1341: Add setting for Support Expansion

AbeFM, 2018-10-18:
### Version
1.40.x

### Operating system type + version
win

### Behavior
When using soluble supports, differential materials or soluble interfaces supports can form rigid solid structures which are very hard to remove. This negates the advantage of the expensive soluble supports.

Consider a vertical collum with horizontal discs. The support grows out into the surround space, and when you try to remove it it self-reinforces. This is generally desirable but many times can be limiting.

In cases like this, it would be preferable to have a lower support inflation so the support can be separated in Z

_Is this a new feature request?_
YES

bubnikv, 2018-10-19:
I do not quite understand. Would you please provide drawings?

AbeFM, 2018-10-19:
Here is a series of curved shelves from a needlessly complex test print. They are supported with the same material as the body of the print, but with a different material interface.
[image]
The above print has no "expansion" so it is not printed beyond the limits of what it is supporting.

Here we see the same part, printed with 3 mm of "expansion", which I've circled some of.
[image]

The far right pictures are colored not by material, but by type, it makes it easier to pick out support against the body of the part.

As you can see, removing the support from the later case is much more difficult - in printing a Christmas Tree, for example, you could end up with the part entirely enclosed by non-soluble support, as I had on my part.

Unless I am missing the setting, I do not see this variable exposed under print settings.

AbeFM, 2018-10-20:
Is this sufficient? Am I happy to generate a better description if I am not being clear.

bubnikv, 2018-10-22:
This explanation is clear, thank you.

The support algorithm by the current Slic3r PE works differently from Cura:
It projects the supports into a grid, which to some extent improves the
stability of supports with the trade-off of higher print time and material
consumption. It is surprisingly difficult to write an optimal support
generator, but we certainly do not give up on the goal.

AbeFM, 2018-10-22:
I assumed this to be the case - the edges of stuff seemed too coincidentally where the next grid would lay.

I don't really expect this to be a trivial problem, and my posted example was chosen for clarity not use-case-justification, but I can assure you many of my prints just come out very poorly this way.

I don't yet have a great suggestion on how to fix it - printing a ring of support matching the outer layer and using that to trim the overshoot could work... But if it seems people agree it's a good idea I'd like to keep collecting ideas here.

bubnikv, 2021-10-05:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2377

morphias2004, 2024-05-10:
Running into an issue directly related to this.

With multi material printing on the XL and using dissimilar materials for supports, this has become more critical.

The support a not generated with a sufficient margin beyond the model for the print to be successful.

The model is effectively printed hard on the outer edge of the supports and will cause print failures.

[image]

[image]

Here you can see the initial perimeters of the first layer on the model have gone off the edge of the support interface layer.

[image]

Increasing the support pattern spacing helps somewhat, but still isn't ideal.

[image]

pavolNovak, 2024-05-10:
Yes please make this new feature. Would greatly improve printing.

morphias2004, 2024-05-10:
The issue is also more prominent in round prints. Square or rectangular prints generate the support sufficiently beyond the model.

[image]

morphias2004, 2024-08-16:
Still an issue.

morphias2004, 2026-09-18:
Is this fixed in PS3, because it's still there in PS 2.9.6?