prusa3d/PrusaSlicer #1342: [Feature Request] Lift Head on Minimum Layer Time (Option)

AbeFM, 2018-10-19:
### Version
1.41.1

### Operating system type + version
Win10

### Behavior
When minimum layer time is reached, the print head will go to minimum speed, after that, there is no way to slow down the print further, or allow the print time/space to cool.

Other slices combat this by lifting the head away from the part to let the time pass (logic of lift_duration = min_layer_time - time_at_min_speed), moving the hot end away from the part and letting it cool.

In practice, people just print a second object, even slowing travel time to exaggerate the effect, but this just leads to waster material and stringing.

_Is this a new feature request?_
YES

Sebastianv650, 2018-10-20:
Does this realy work as desired? I know Cura has this feature, but I would expect too see bits of oozing all over the print if this is triggered. When you lift the head for some seconds, there will we some mm of oozed material which has to go somewhere.

AbeFM, 2018-10-20:
Yes, it can work well. It is good as an option, I agree there are cases
where it is challenging but there are also cases where it is enabling.

I could do a side by side in some other packages if people would like to
see a comparison.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

bubnikv, 2018-10-22:
It would certainly be useful with a tiny wipe tower to prime the extruder
after the wait.

AbeFM, 2018-10-22:
I've used this feature >20 times in other software packages. I find the issues you raise are not a problem in practice - for instance, if the minimum layer time is 5 seconds, and your print is 2, the wait time is pretty short.

It's not to say they aren't considerations, but the net effect is enabling prints which don't work otherwise, I get strings and waste time and filament jumping to a second object that is only there to waste time. And you can't find tune the time wasted.

If you cannot download Ultimaker's CURA and try the setting in practice, send me an STL and I can hit slice for you to assuage your fears.

szafran81, 2018-10-23:
+1

One can even wipe into infill - without the need for a wipe tower. Just go to the middle of the print after the wait time and purge there a bit of filament, and then resume printing.

AbeFM, 2018-10-23:
A possibility. But typically when you're into minimum times, there's precious little infill to choose from.

Likewise, you may leave the string on the outer wall while approaching the center. I don't think it's needed but Bubnikv's tiny tower might actually be better?

Either way, the waiting really does work. https://www.thingiverse.com/thing:3153356 is a case where it would be handy - picture it upside down. :-P

szafran81, 2018-10-23:
I often print those with something larger. But then If I have something small and don't want to wait or add unnecessary big parts, then waiting would be a blessing.

AbeFM, 2018-10-23:
A better example may be an Eiffel Tower, it starts off with long layers,
but would require a large unutilized purge tower or sub-optimal settings
for the lowest 95% to still achieve a nice top.

Nalha, 2021-04-02:
I did this by adding this G-code to before layer change, pauses on every level after 11.8mm Z.

```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
{if layer_z > 11.8}; wait
G1 Z{layer_z + 5} E-3.20000
G4 S10
G1 Z[layer_z] E3.20000{endif}
```

There is a small amount of plastic left on the print but because I raise straight up and down again it is inserted into the hot end and has never left any marks. Picture of printer while paused https://i.imgur.com/Lr9hk2L.jpg

BluPix, 2022-01-16:
Hi, I have another idea on how to handle this. If the print contains an infill, would it be possible to find a spot in the middle of the infill and wait there without having to move the Z axis? And if the layer does not contain infill, try not to slow down the outer perimeter.

Nalha, 2022-01-16:
> Hi, I have another idea on how to handle this. If the layer contains an infill, would it be possible to find a spot in the middle of the infill and wait there without having to move the Z axis? And if the layer does not contain infill, try not to slow down the outer perimeter.

The problem with waiting close to printed material is heat buildup, that part of infill will warp and possibly collapse.

batata004, 2022-01-16:
@yoyo42 is still the best solution so far, the simplest and most effective.

yoyo42, 2022-01-23:
Since the actual meat of that idea isn't in this thread, here's what I said:

_Having another distant tower works but is wasteful. However the only other realistic option I can see would be to ensure that after moving away and waiting and possibly oozing a little, the return is given extra Z lift and also always restarts in an internal infill area so the ooze can wipe off in a hidden area rather than anywhere visible._

Tupson444, 2022-10-07:
A while ago I tried doing this manually in G-code and failed. I wanted to print just a few small objects, which I usually print many at once in order to increase layer time for better cooling. The object is really small near the top (~2mm diameter) with a significant overhang. That's why I wanted to move the nozzle away for a couple of seconds after each layer.

I'll try again to do it manually, inspired by @Nalha 's suggestion. However, it would be more convenient to have it as an option in the slicer. Even though it isn't perfect regarding oozing, it would still be OK in many cases.

MC42, 2024-03-13:
(The above mentioned issue that was closed was mine, must've missed it when I went looking for the variable a few months back.  Sorry for missing those issues!)

Adding on to what others have said, this feature would be super helpful to be built-in for Cocoa Press users.  Especially since Cura seems to have this feature as well.  

Alternately, a variable to hack it in as layer change GCode would be workable too, just something to not make it entirely manual would be ideal.

Thank you in advance!

pilot51, 2024-04-16:
I tried @Nalha's solution (from [Reddit](https://www.reddit.com/r/prusa3d/comments/mimek3/replacing_autocooling_with_pause_between_layers/) before I found this issue) because I had a print that was going to spend the last several layers on a very small part (a single valve stem cap). It worked great! There was a tiny bit of stringing when it lifted, but that disappeared when it dropped back down.

I'm mostly printing in PHA, which is kind of hard to get right. I find that it takes longer to solidify than PLA and it warps very easily as the higher layers contract. So far, with the pause between layers (I did up to 15 seconds), I've had no problems.

I've had PLA and ABS prints turn out badly or even fail because of this layer time issue. Slowing it down doesn't really help when the area being printed is barely 1cm² in size and each layer still takes no more than a few seconds.

I like the idea others have mentioned about a cooling setting for a minimum layer time, inserting a pause that brings the previous layer time up to the minimum.

batata004, 2024-04-16:
I cant believe prusa still didnt implement this. Prusa is losing to competition in hardware (Bambu labs is taking the market that prusa failed to take because of lack of inovation and terrible price policy) and also in software. Such a simple solution, to lift the head when the layer printing time is too low, is incredible simple to implement to a company that makes milions of dollars and lots of people would benefit with it.

I will let my 2 cents: if you can, sell your prusa and buy a bambu labs. That's a much much much better printer, much much much better print time and much better quality... and did I say it's a lot cheaper?

wafflecart, 2024-04-16:
> I cant believe prusa still didnt implement this. Prusa is losing to competition in hardware (Bambu labs is taking the market that prusa failed to take because of lack of inovation and terrible price [...]

Nobody asked, stop spamming. If you think it’s simple we await your pull request with the code..

batata004, 2024-04-17:
> Nobody asked, stop spamming. If you think it’s simple we await your pull request with the code..

Nobody asked your opinion too. Stop spamming other people's comment. If prusa sucks and lots of people dont like it anymore, every single person has the right to a opinion.

Regarding the pull request you asked me, you can wait seated because I will never contribute to anything from prusa anymore. Overpriced, old tech, with slow printing and zero innovation in the last 3 years.

pilot51, 2024-04-17:
This is not the place to argue about brand. I have my reasons for choosing Prusa over more popular brands. I spent months making up my mind and my decision is firm. I love my printer despite this little issue in the slicer and you're trying to take away the joy. It's clear you no longer use PrusaSlicer and don't intend to use it in the future. This issue does not affect you. So please stop trolling and unsubscribe from the issue.

batata004, 2024-04-17:
Unfortunatelly I still have a prusa that I cant sell because it has 2 issues that nobody will buy it because of those. So despite owning a bambo lab I still have this terrible printer. Solving this feature request and maybe many more, will help me to sell this terrible product that I overpaid.

AbeFM, 2024-04-17:
Uh huh. I'm not even fixing my Prusa. First since it's easier and cheaper
to buy the other machine, and second, hours of maintenance pet year and
number of failed prints.

As the other poster mentioned, this is not a big lift. They could upgrade
cooling, though software is cheaper... And they can get the code straight
from CURA who has had this for maybe a decade?
 3 friends bought A1 Bambus after seeing mine. Plus two who bought X1's
before the A's were a thing. After seeing the Prusa..friends bought Enders
back in the day. I'll plainly admit I was pretty excited about the XL, but
it took years, not minutes, to get MMU to kinda work. Get stuff working,
then sell it. I'd consider it if most people said it printed well and it
cost $1000 less.

But fingers in your ears guys. Good luck with that.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

wafflecart, 2024-04-17:
> Nobody asked your opinion too. Stop spamming other people's comment. If prusa sucks and lots of people dont like it anymore, every single person has the right to a opinion.
>
> Regarding the pull [...]

Yes everyone has right to an opinion but this is github and you are leaving an unprovoked comment in a feature request from 2018.. just out of nowhere blasting an entire company for an albeit minor feature. The amount of people that would not buy a printer because of this feature request is VERY SMALL and as a result it probably hasn't been a high priority to be implemented.

As others have said this is not the place to argue about brand X over brand Y, go to twitter or reddit or some other social media.

samuelcomeau6, 2025-03-17:
Cura has this feature. Also with the core One this feature needs more attention. Reviews have it performing worse in PLA prints than a mark 4 because of too much heat. This would go a long way towards mitigating the issues.

closed 2026-09-29 by prusaslicer-bot[bot]