prusa3d/PrusaSlicer #1344: Generating Support - Contact Z distance too big

borrisbecker, 2018-10-20:
### Version
1.41.1-beta+

### Operating system type + version
10.13.6 (17G65)

### Behavior
The distance between the support and the object is that of the nozzle diameter and not the "contact z distance". When printing large horizontal overhangs, there is not enough contact and the print fails.
Contact z distance needs to be as per setting (0 or 0.2), not the nozzle witdh.

#### STL/Config (.ZIP) where problem occurs
[Contact z distance.zip](https://github.com/prusa3d/Slic3r/files/2498244/Contact.z.distance.zip)

Sebastianv650, 2018-10-20:
This was already discussed in #102, together with a work around how to get the real, wanted gap setting.
So far this "theory" seems to be not very popular on the programmers side, but they are reworking the support system so fingers crossed we will get this one day.

borrisbecker, 2018-10-20:
Thanks Sebastian

I’ll take a look at this, I’ll delete / close my thread when I get the chance. 

Cheers

Chris 

Chris Lloyd | chris@chrislloyd.com | www.chrislloyd.com | FreeformSpeakers | +44 7739 117670

bubnikv, 2018-10-22:
Your issue is a duplicate of Supports not generating interface layers #1306
This is a regression issue found in 1.41.1-beta and 1.41.1-rc, introduced with the reduction of the support interface regions. Fixed with eb59568
It will be part of the 1.41.1 final release. Closing.

closed 2018-10-22 by bubnikv

bubnikv, 2018-10-22:
Your issue is a duplicate of Supports not generating interface layers #1306
This is a regression issue found in 1.41.1-beta and 1.41.1-rc, introduced
with the reduction of the support interface regions. Fixed with eb59568
<https://github.com/prusa3d/Slic3r/commit/eb595683680e78b60ae00038f877927033509413>

It will be part of the 1.41.1 final release. Closing.