prusa3d/PrusaSlicer #1345: GCODE send for MMU2 in single extruder mode to octoprint contains problematic T? gcode

ruedli, 2018-10-20:
### Version
_Version of Slic3r used goes here_
v1.41.0
_Use `About->About Slic3r` for release versions_
v1.41.0+win64
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
stable
### Operating system type + version
_What OS are you using, and state any version #s_
windows 10
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
Any gcode generated for the MMU in single material mode, contains a T? gcode. The standard profile in fact contains this:

; select extruder
T?

Whereas this is fine when printing from the SD card, when such a file is send to octoprint, it cannot be used. From SD card, there is a prompt which filament to use, but when using octoprint  there is no prompt, instead it attempts to load filament from extruder 2, which fails when no filament is present.

Until the MK3 / MMU2 firmware deals with this GCODE properly, gcode T? should never be present in gcode sent to octoprint. When I manually changed the code from T? to T0, it worked, however, not enough material was extruded, so more gcodes might be to purge extra material after loading.

_Is this a new feature request?_

Yes, until the MMU2 firmware deals with it more graciously. Allthough is is easy to fix ones you know what is causing this, I imagine many new MMU2 users with an octoprint server struggeling with this one....The numerous failing prints frustrated me...

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
It is not .stl specific, the problematic gcode sits in the profile, in the tab "printer settings" setting: "custom Gcode" section "Start G-code", the relevant section is this (copied from Start G-code):

M107
M115 U3.4.0 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling

G21 ; set units to millimeters

;go outside print area
G1 Y-3.0 F1000.0
G1 Z0.4 F1000.0
; select extruder
T?                                                          <===== THIS CAUSES THE PROBLEM
; purge line
G1 X55.0 E8.0 F2000.0
G1 Z0.3 F1000.0
G92 E0.0

bubnikv, 2018-10-22:
> Until the MK3 / MMU2 firmware deals with this GCODE properly, gcode T?
should never be present in gcode sent to octoprint.

It is not the MMU firmware that has issues with T?, it is OctoPrint. Gina
(the author of OctoPrint) suggested to us to write an OctoPrint plugin to
handle these cases.

For now, we recommend to you to change the T? to T1/2/3/4/5 in your G-code,
or to change it inside your start G-code of the "MMU Single" profile (you
will need to save the modified profile under a different name than the
system one), or to print with the multi material MMU profile. Does it solve
your issues?

ruedli, 2018-10-22:
Thanks for looking into this.
I already updated the MMU2 single extruder profile and updated it to use
T0, T1, T2, T3, T4. It uses the configured extruder as now present in the
GCODE, so no dynamic "changing the extruder".

As for the problem: I do not think it is a good idea to wait until a plugin
has been written by Prusa and have all the octoprint users "fail" with this
profile. Remember, Prusa even recommends building in a Raspberry Zero W in
their MK3, and offers a "prepared" octoprint image in their support pages.

As for the solution,
I just entered the T? command from the terminal tab of the octoprint web
page of the octoprint server connected to my printer.
This results in the prompt on the display and after selecting the extruder
it loads. Furthermore the printer goes in Pause mode and returns after
loading filament.
(By the way, this also works with a cold ! extruder and does some grinding
of the filament. This should not be possible in my opinion.)
As mentioned, T0,1,2,3,4 when present in a file work as well. T? in a file
does not work.
Is this because T? is handled as a special GCODE, when present in a file???
By Octoprint? Would be great if Gina could confirm that. If so, Prusa would
better handle the "select extruder" by issuing a Tx command, or e.g  T9,
knowing that we only have T0 to 4 for selecting valifd extruders. At least,
the T9 would reach the printer, also from a file.

Whether this is a solution, depends on the details that Gina provided and I
can only guess. Is the T? matched as part of a regular expression and
substituted by octoprint?

As for the plugin:
Now the question is why a plugin needs to be written and what this plugin
should do. T? through USB by means of octoprint from the terminal seems to
work.

The plugin should be an user-interface to select the "behaviour" of
octoprint for translating T? commands, and substituting them into
T0,1,2,3,4... This might work and could be a solution.
It is too dedicated to the Prusa MMU2 for Gina to write such a plugin, so
her suggestion to Prusa to write it makes sense.

My suggestion is to at least do something about it, so that it does not
fail, like putting a T0 as default in the standard profile. Using T9 in
MK3/MMU firmware in stead of T? seems the best way out long term. The
(future) plugin depends on too many things that can go wrong, like people
not installing the plugin or configuring it correctly. Also, form an
architecture point of view it seems bad to need to install a plugin to make
the Prusa MK3/MMU single extruder profile work with octoprint. That would
be a first case and I think it looks bad. It is a relatively simple
solution as well, in fact I think I could patch the c code of the MMU to
make it work that way, for your information this is VERY SIMPLE:

In Marlin_main.cpp of the current 1.02 firmware with the MK3 you see in
line 6828:

if (*(strchr_pointer + index) == '?') {

replace this with:

 if (*(strchr_pointer + index) == '9') {

Further the check for a valid gcode should also be changed, this is line
6824:

      if ((*(strchr_pointer + index) < '0' || *(strchr_pointer + index) >
'4') && *(strchr_pointer + index) != '?') {

replace it with:

      if ((*(strchr_pointer + index) < '0' || *(strchr_pointer + index) >
'4') && *(strchr_pointer + index) != '9') {

Finally the documentation and test cases need to be updated, but I did not
look for those.

So... I do not entirely agree. Hope you can understand my motives, they
were written from the perspective of what I feel are the cleanest solution.
A plugin seems simply to cumbersome and exotic, for something as simple as
this.

Cheers Ruud

ruedli, 2018-10-22:
and, finally, specifically for the single material usage, there is a line
of code to call "mmu_load_to_nozzle();" so also change:

if (*(strchr_pointer + index) == '?')// for single material usage with mmu

to

if (*(strchr_pointer + index) == '9')// for single material usage with mmu

Sorry for missing that one... strange that the '?' is not coming through in
Octoprint, there are other command that also use this (e.g. PINDA
temperature compensation offsets)

kind regards,

ruud

ruedli, 2018-10-23:
_and_ to make it more interesting,

- I _did_ download the firmware,
- made my "T9" change instead of "T?" in the three lines mentioned,
- uploaded the newly build firmware into my MK3,
- updated the profile to use T9 instead of T?
- uploaded GCODE customizatition in Slic3r  to use this T9

And.... By printing though octoprint, I got a prompt on the MK3 as
intended..... Yessssss ;-))))

So... I think it is MUCH easier for Prusa to work around the T? handling of
Octorpint, by using T9 instead. When delving through the firmware for the
MK3, I notcided there is a lot hardcoded anyhow, e.g counters of extruders
are now hardcoded as 0..4, whereas the MMU1 code has hardcoded counters
0..3 instead of setting and using the #define EXTRUDERS.... (which seems
only used in the original Marlin code...)

Anyhow, I learned a lot, I disagree now strongly to writing a plugin,
instead I would update the T command using the T0, T1, T2, T3, T4, and
T9... in which case no changes whatsoever are needed in Octoprint or Prus
MMU2 specific patches of Gcode...
(which if you _really_ wanted to are still possible, by now catching T9
instead of T?)

Cheers Ruud

ruedli, 2018-10-23:
Now that a clean fix exists by doing thing differently in firmware, should it be transferred to the firmware release????

bubnikv, 2018-10-23:
I will talk to the firmware guys. Thanks for your investigation.
Maybe we may use T99 or maybe T9999 to be future safe?

ruedli, 2018-10-23:
OK, so here are some more firmware update try-outs... After being successful implementing 'T9', I tried with a 'TX', whereas this worked from the terminal, it would not be streamed by Octoprint, much like a T? therefore.

I then reverted to using T9 and implemented support for both T? and T9. If you implement this straightforward, the firmware is 30 bytes larger, but by introducing a tmp variabel, I could reduce this and make it 14 bytes smaller again...

Whereas it indicates a problem in Octoprint, or perhaps a plugin of Octoprint, I do think it is worth "patching" by Prusa. You simply do not want users of Octoprint hitting this. Note that I also found another user in the Octoprint forum that stumbled over this and was confused.

The change I propose is downwards compatible, so T? in gcode on SD cards would still work. For it to work best, also the printer profile GCODE in Slic3r needs to reflect "T9", 

My proposed code now looks like this replace line 6818-6855 with:

  else if(code_seen('T'))
  {
      int index;
      st_synchronize();
      for (index = 1; *(strchr_pointer + index) == ' ' || *(strchr_pointer + index) == '\t'; index++);
      
      //RR, support also the T9 command in addition to T?. T? has problems with some Octoprint installations.
      //RR, introduce a variabele to remember that a user wants to see the menu for selecting the extruder
      boolean tmp_user_selects_extruder = (*(strchr_pointer + index) == '9') || (*(strchr_pointer + index) == '?');
      
      if ((*(strchr_pointer + index) < '0' || *(strchr_pointer + index) > '4') && ! tmp_user_selects_extruder) {
          SERIAL_ECHOLNPGM("Invalid T code.");
      }
      else {
          if (tmp_user_selects_extruder){
              tmp_extruder = choose_extruder_menu();
          }
          else {
              tmp_extruder = code_value();
          }
          snmm_filaments_used |= (1 << tmp_extruder); //for stop print

          if (mmu_enabled)
          {
              mmu_command(MMU_CMD_T0 + tmp_extruder);

              manage_response(true, true);
              mmu_command(MMU_CMD_C0);
              mmu_extruder = tmp_extruder; //filament change is finished

              if (tmp_user_selects_extruder)// for single material usage with mmu
              {
                  mmu_load_to_nozzle();
              }
          }
          else
          {
#ifdef SNMM

bubnikv, 2018-10-23:
I discussed the topic quickly with Pavel, our firmware lead. We may define
a separate M code to be equivalent to T?. The T? was not really a good idea.

ruedli, 2018-10-23:
I couldn't agree more! It sort-off stood out from the rest of the processing... ;-) 

In the mean time I'll enjoy my private little T9 patch ;-)

PavelSindler, 2018-10-23:
Using TX or similar seems like the best option from my point of view. We will solve this issue in next firmware release.

ruedli, 2018-10-23:
Hi Pavel,
It should indeed be the -or similar-, as TX behaves exactly the same as T?.

T9 _did_ work OK for me however, both terminal and streaming.

eboston, 2018-10-24:
> Using TX or similar seems like the best option from my point of view. We will solve this issue in next firmware release.

Both T? and TX violate the standard GCode format as I see it.  The letter needs to be followed by a number.  A M code is the better way to implement this.  The printer has only one tool (T0), but an external mechanism for controlling the filament going into the tool.  Control for this is what the M codes are for.

ruedli, 2018-11-09:
> Using TX or similar seems like the best option from my point of view. We will solve this issue in next firmware release.

NEW important information:

The issue was not tied to T?, but to the processing of the first T gcode when streamed.

when I found that it did not interpret:
- In a multicolor gcode file, the FIRST T command
- in a single material MMU gcode issues with the first (and only) command

I went back to the stock firmware and added an additional T0, BEFORE the actual T command (either T[initial_layer] or T?) in my profile under GCODE when print starts.

This worked as a workaround and I did not need a patched firmware to have proper processing of the (now second) T? command

Therefore the concentration of where the issue is should be in my opinion not on T? but on under which circumstances the first T command gets dysfunctional.

Hope this additional information is useful, it significantly impacts the hypotheses of the likely cause of the problem.

TimothyGold, 2018-12-20:
All of this g code seems to work fine in octoprint when I first send the print right after opening slic3r. If I change from single mode profile to mmu mode and then back again the filament selection on the mk3 lcd never shows up and defaults to t0. I have to close slic3r and reopen it and then reslice the same single mode project to get the filament selection to show up again on the mk3 display.  Just fyi I am working with the very latest version of slic3r and firmware.

bubnikv, 2018-12-20:
I have a difficulty to understand your issue. Are you saying, that in the
first case the G-code generated by Slic3r is bad, and in the second case
(after you close Slic3r, start it again and slice) the G-code is correct?
Would you please attach the two G-codes? Thanks.

ruedli, 2018-12-20:
Dear bubnikv,

The issue description is old and obsolete now. It referred to gcode T? that
should result in bringing up a menu to select which extruder to use for the
MMU2 in single print mode. In my case it didn't.

The pre-3.5.0 MK3 firmware had a problem with its cache handling, Under
some circumstances T? would work, under other circumstances it didn't,
especially when streaming GCODE commands, like e.g. from Octoprint. My
suggestion was to not use problematic code in Slic3r, however, ALSO having
two, equal T? commands in the GCODE initialization then helped for me. The
issue was addressed in the i3 MK3 firmware 3.5.0 by the Prusa firmware
team. (fixing this problem did not need code in the MMU2 unit)

Hindsight: referring to the problem as it was (at that time) "newest"
firmware of the printer was unfortunate, should have quoted the firmware
version that was not working for me. The problem I refer to is fixed as of
firmware v3.5.0 for the i3 MK3.

Hope this clarifies it for you.

Ruud

TimothyGold, 2018-12-20:
@bubnikv 
I think I made need to start a new issue then. I am still having unreliable issues with this filament selector option showing up on the mk3 when starting a print from octoprint.   Last night I started a print from octoprint and the option to choose which filament was offered. I chose filament 5 and the print progressed. My daughter accidentally shut off the printer during the night and the print failed. I used the same gcode that included the latest Tx and Tc options which worked the first go around. This time the option to select the filament was never offered and it started the print saying it was going to use F1.   I had to quickly move the filament to position 1.  I believe this is probably a firmware issue because the gcode didn't change from the slic3r. 

Firmware v3.5.0
Slic3r version is the latest with default prusa single mode profiles using the latest Tx and Tc for filament selection. 
Octoprint is also the very latest release. 

Every single mode print I start seems to have some issue with reliably being the filament selection at the beginning of the print. Most of the time it defaults to F1.

ruedli, 2018-12-20:
@Tim Gold,

I also noticed it behave wrongly recently once with the firmware 3.5.0. For
me, what worked was to ensure the MK3 was totally ready with its boot
sequence before starting a print. When I start the printer, often Octoprint
winds up in a state that it is not connected. As I connect Octoprint to the
printer, the printer then reboots. When I then start a print too soon, I do
not see the Tx menu. Could this be the cause of what you see? In that case,
just wait long enough until the MMU2 went through all its startup
movements.

This behavior was different though from the "caching" problem.

Regardless of anything, before reporting a possible problem, it is worth
flashing 3.5.1 MK3 and 1.0.3 MMU2, as they are now current and recently
released.

Kind regards,
Ruud

TimothyGold, 2018-12-20:
@ruedli 
Yes I think this might be what I am experiencing. I did upgrade the mmu2 to 1.0.3 and I think this may have made this problem worse because with this new firmware for mmu2 the homing sequence doesn't happen on boot anymore. It now happens when the filament load command is sent. I haven't upgraded to 3.5.1 yet. I will do that and then try your workaround. One more thing to add. I just tried to start a print again probably not waiting long enough after connection but this time it started using f1 but never fully loaded the filament and started printing empty. I don't have the sensor on so it ddidn't ense the filament missing.   Anyway I do think there are some bugs with this process that show up with certain circumstances. Thanks for your input.

ruedli, 2018-12-20:
You are welcome, please let us know after trying if it works correctly in
your setup after waiting with starting the print until the boot sequence is
completely finished.