prusa3d/PrusaSlicer #1346: Support blocker not working on this model

tanguym, 2018-10-21:
### Version
Slic3r 1.41.1-rc

### Operating system type + version
Windows 10 x64

### Behavior
I tried using a support blocker to remove an unneeded support structure on an stl file, but it still generates the support. As you can see, the support below the eye is not required, while I do want support in the eye. So I added a blocker for it. But it seems the blocker is ignored. The stl file I used is https://www.thingiverse.com/thing:3148236
[image]
[image]
[image]

_Is this a new feature request?_
no

#### STL/Config (.ZIP) where problem occurs
[Skullpkin.zip](https://github.com/prusa3d/Slic3r/files/2498903/Skullpkin.zip)
[config.zip](https://github.com/prusa3d/Slic3r/files/2498904/config.zip)

bubnikv, 2018-10-22:
What you see is an effect of how the support generator works: It first
calculates columns as polygons in successive 2D layers, and first in the
final step these polygons are hatched with the support extrusions. The
current algorithm believes the column you encircled supports the column
above, it is not aware, that piece of the column above is so thin that no
extrusion line fits there.

We need to improve the support algorithm, though the case you shown is
relatively harmless.

bubnikv, 2021-03-23:
Fixed with af9c7c967f08fcf22ffa628c324c90eda65230f1

    Implementing a new switch for the shape of support towers:
    expanded to a grid (the old way) vs.
    snug (like the upstream Slic3r, Cura or Ideamaker).
    
    Snug supports suffered from the degeneracies when merging overhang islands
    over a large number of layers when projecting the support towers down.
    We borrowed the idea & a bit of code from Cura by simplifying the support
    polygons by closing the concave cracks, see the smooth_outward() function
    and the MutablePolygon class.

closed 2021-03-23 by bubnikv