prusa3d/PrusaSlicer #1347: How to use Prusa Slic3r Settings specific configurations from command line?

tobymurray, 2018-10-21:
### Version
1.41.0+linux64

### Operating system type + version
Ubuntu 18.04

### Question
How does a user load the specific filament/machine configurations on command line?

The files available at [Slic3r-settings](https://github.com/prusa3d/Slic3r-settings) appear to have a whole bunch of different configurations - different printers, different filaments, etc. Looking at the [Slic3r documentation](http://manual.slic3r.org/advanced/command-line) and the command line help (`slic3r --help`) I see that the `--load` flag can be used to load configuration:
```
--load <file>       Load configuration from the specified file. It can be used 
                    more than once to load options from multiple files.
```
This "works" to load the `.ini` file, but it's not clear how to select things (like in the UI). For example, what is needed to "select" the Prusa PLA as the filament and the MK3 as the printer? Is this `.ini` file specifically a Slic3r PE GUI configuration file, or can it be used in the same capacity from the command line?

`slic3r --load ../Slic3r-settings/live/PrusaResearch/0.2.5.ini --save test.ini` for example shows what appears to be a unified configuration with e.g. `filament_cost = 0` (presumably a default value) instead of any of the configured filament costs.

_Is this a new feature request?_
No

bubnikv, 2018-12-12:
This is a good question. --load is to load a config file, not a config bundle. The Prusa's system profile is an extended form of a config bundle, so you should not use a --load.

However, Slic3r PE allows loading a bundle as a config file, at least from the UI. In that case it activates the profiles, which are marked as active in the config bundle file. I did not test it from the command line though.

For offline slicing, you may try to load a 3mf file, which you previously exported, but I suppose this will not help you.

I agree with you, that it would be nice to be able to activate installed profiles from the command line, and I suppose this is the feature you are asking for.

tobymurray, 2018-12-12:
If the functionality does not currently exist, then agreed - I guess this is a feature request. Hopefully the code already exists to drive the UI?

Ultimately what I want is to slice from terminal without maintaining a separate-but-equivalent configuration from what I use in the UI, and I'm not seeing how to do that without pre-processing the file to build a single configuration. E.g. when I update a value in the UI and save it (change the temperature for a specific filament) I'd like to be able to use that change from the terminal.

> you may try to load a 3mf file, which you previously exported, but I suppose this will not help you.

I'm not familiar with 3MF, I can take a look though.

bubnikv, 2019-05-24:
We currently process the GUI profiles with a bit of code, which due to Perl legacy (combining C++ and Perl GUI code) contains both the business logic and GUI updates. We shall split the two and push the profiles code into command line only libslic3r library. Then it will be easy to support selecting the GUI profiles from the command line.

Agadicz, 2022-08-27:
hi is possible to use config bundle in command line version 2.4.2?   I try it but without success... Thanks for the answer

closed 2026-09-29 by prusaslicer-bot[bot]