prusa3d/PrusaSlicer #1349: Slic3r hang up while splitting part

Massa0815, 2018-10-23:
### Version
1.4.1 precompiled binary 64 bit fpr Windows

_Use `About->About Slic3r` for release versions_

### Operating system type + version
Windows 7

### Behavior
load the attached File and try to split. An Error occures

#### STL/Config (.ZIP) where problem occurs

[Halter_seperat_abger.zip](https://github.com/prusa3d/Slic3r/files/2505128/Halter_seperat_abger.zip)

Debugging Code:

Problemsignatur:
  Problemereignisname:                       APPCRASH
  Anwendungsname:                            slic3r.exe
  Anwendungsversion:                         1.41.1.0
  Anwendungszeitstempel:                  5bb76a35
  Fehlermodulname:                             XS.dll
  Fehlermodulversion:                          0.0.0.0
  Fehlermodulzeitstempel:                   5bb769f3
  Ausnahmecode:                                  c0000005
  Ausnahmeoffset:                                00000000001cae42
  Betriebsystemversion:                        6.1.7601.2.1.0.256.48
  Gebietsschema-ID:                             1031
  Zusatzinformation 1:                          49d6
  Zusatzinformation 2:                          49d633c555d53ca3dabfb4b420b67f0c
  Zusatzinformation 3:                          df01
  Zusatzinformation 4:                          df01757264f8d70c9bc29b9d4d47d2eb

FidelCapo, 2018-10-23:
Thank you for report. I can reproduce this bug on Win.

FidelCapo, 2018-12-07:
This issue should be fixed in the next release.

bubnikv, 2019-02-05:
The crash now is fixed on master by swapping the triangulation library with a proven glu tesselator (written in 1994 in plain C by SGI). The fix will be part of the Slic3r 1.42.0-alpha6.

While the triangulator does not crash, if the input mesh is broken, the cut is not always triangulated. Namely, inverted parts are not triangulated.

Closing.

closed 2019-02-05 by bubnikv