prusa3d/PrusaSlicer #1350: feature request - auto plate filling

tkapler, 2018-10-23:
I often print something small like e.g. Lego bricks or some badges and i need to print as many as possible - to fullfill the plate. And from what i see at Prusa lab, i am not the only one, there is almost always at least one printer printing such project.

Today, there is only one way how to do it - select object, guess what will be the total number and click Set number of copies and then Arrange, but it has many disadventages:
a) if it too much, you than have to click - as many times as it will finaly be the right number, if it is too little, i need to click + many times and then again Arrange to test if it fits. 
b) the width of the space between the object with Arrange is fixed, so for small objects like lego brick it can be even bigger than the object iself so only third or half of area is filled.
c) it always check just one X,Y orientation of the object, sometimes just rotating 90° will end in fitting much more objects, or e.g. rotating one half one way and the other half the other way
d) with more then one objects (e.g. object cutted in two) is is even more difficult.
...

So there could be at least
a) option for Arrange to set the space between the objects
b) auto plate filling - at least the same stupid one as current, with just removing the copies that do not fit on the plate, more optimaly with some added options like checking if rotation of the object would save some space (see https://docs.opencv.org/3.4/d3/dc0/group__imgproc__shape.html#ga3d476a3417130ae5154aea421ca7ead9), and maybe 90° rotation of some of the objects, so that it would better fit. The best would be some advanced algorithms you use for the laser cutter.

supermerill, 2018-10-29:
>a) option for Arrange to set the space between the objects

An option exist: "duplicate_distance" or "Distance between copies" (default to 6), but it seems it's not added inside the gui.

lukasmatena, 2019-11-10:
a) You can just right-click the object, set number of instances to something that's too much and hit arrange. All that does not fit will be arranged on a virtual bed to the right of the real bed, so you can easily delete the extras by rectangular selection. It is much less work to do it that way. I'm not the author of the arrange function (@tamasmeszaros is), I bet this is exactly what the auto-fill-bed function would have to do. Whether this should be implemented and what other pitfalls there are I don't know.

tkapler, 2019-11-11:
@lukasmatena you are right that this is now a hack, but still a suboptimal hack and does not work for multiple objects on the bed, when you want the same number of the object on the bed, as you have no control what will be on the bed and what elsewhere.

closed 2020-11-20 by tamasmeszaros

bubnikv, 2020-11-21:
It will be implemented in PrusaSlicer 2.3.0-alpha4. Closing.

commit adf81af6 (2020-11-20): Prototype feature: fill plater with instances of selected object

#fixes #1350