prusa3d/PrusaSlicer #1351: support for a 3Dconnexion mouse

ruedli, 2018-10-23:
After working with a 3D mouse, from 3Dconnexion in CAD and some other packages, I have to say that this works really nice compared to zooming and navigating with a normal mouse.

Sli3r could benift, not only for "moving around", but also placing and rotating objects. These could then be snapped to the lowest most horizontal surface 

_Is this a new feature request?_
Yes

szafran81, 2018-10-23:
+1 for SpaceMouse

mgg4, 2019-01-12:
Please, yes.  This would be a great improvement to our workflows.

sfmarkham, 2019-02-16:
+1
[Give Your Users Best-in-Class 3D navigation](https://www.3dconnexion.com/service/software-developer.html)
[Confused: What API to use to add 3D Mouse support to our app](https://www.3dconnexion.com/forum/viewtopic.php?t=3651)

ruedli, 2019-03-21:
> +1
> [Give Your Users Best-in-Class 3D navigation](https://www.3dconnexion.com/service/software-developer.html)
> [Confused: What API to use to add 3D Mouse support to our app](https://www.3dconnexion.com/forum/viewtopic.php?t=3651)

The 3Dconnexion website offers free access (on their developer's forum) to SDK's for windows, mac, unix/linux:

SDK Download
Support Resources
3Dconnexion developer's forum
Email support for Windows: apisupport-win(at)3dx.com
Email support for Mac: apisupport-mac(at)3dx.com
Email support for UNIX/Linux: apisupport-unixlinux(at)3dx.com

srobert910, 2019-03-21:
+1 for spacemouse support.

bubnikv, 2019-03-22:
I wrote support for the 3D mouse once for a Win32 application. It was not
difficult indeed. but to be usable the refresh rate of the 3d scene has to
be quite high. Also there are currently around 600 issues open, many more
important than this one.

cggorman, 2019-05-26:
Adding my support for this feature request!

After more than 10 years using 3D mice, it's quite difficult to go without.

ruedli, 2019-05-26:
Yeah, know what you mean ;-)

I got the feedback that Prusa first wants to resolve issues and more
generally used feature: this one has a low opportunity.

I tried locating the place in the code where this might need adding. Sofar
I manged to at least set up an environment to compile it, but I could not
see where to elegantly fit it in. If someone knowing how the mouse
navigation (panning / zooming / rotating) os located can give hints, I
could try further: apparently the coding is not too difficult: just knowing
where to do it.... kept me from applying the change.

Kachidoki2807, 2019-07-05:
I have to add my vote to this feature request. There is nothing more disturbing after spending an hour or two from the CAD software to have to switch your brain from the left hand to the right hand regarding the camera management.

ruedli, 2019-07-15:
Yes... happens to me a lot: Fusion360 on the left monitor, PrusaSlicer to the right, and then you wonder why nothing is moving on the right when you zoom and rotate intuitively... Especially when manipulating support blockers/enforcers and modifiers, the mouse positioning is not that great. This needs a "special move", so that the object is not rotating, but just moved in 3D. The rotating/zooming "inputs" could perhaps be used to move the view?

workinghard, 2019-08-05:
+1. Would be great to have it!

cybie2k, 2019-09-04:
+1 This would be huge time saver for me.

JakeHertenstein, 2019-09-04:
Maybe if we all post duplicate Feature Requests we'll get more attention

workinghard, 2019-09-04:
Don't think so. Devs are most likely busy with all SLA feature lists.

Kachidoki2807, 2019-09-05:
I would say, this feature also concerns people whom use the SLA tech. And I assume it's even more important for 'em to "navigate" fluently through the 3D scene to check the tree supports.

MiniZagGi, 2019-09-25:
+1

Maybe even support for the Enterprise version as well. I would love to map bottoms to it and see the icons on the screen.

Kachidoki2807, 2019-09-27:
Just saw the Jo's video on twitter, yesss. :-D

prowlmedia, 2019-09-27:
> Just saw the Jo's video on twitter, yesss. :-D

what video?

JakeHertenstein, 2019-09-27:
https://twitter.com/josefprusa/status/1177531505301696512

It's alive!

prowlmedia, 2019-09-27:
Cool I suppose this thread can be closed now :) Whoop

ruedli, 2019-09-27:
Well, the tweet looks promising, but the thread should be closed when there
IS support in PrusaSlicer release...

Good to see, that it gets some attention now, it really is more than "just"
nice to have, given the MANY people confirming their expectation (and
hopes) to be able to use their 3D mouse!

vojtechkral, 2019-09-30:
@probonopd Do you know whether we can assume that all linux distros have `libudev`? (Context: I'm looking into building/linking/distributing the 3D connexion support on Linux.)

probonopd, 2019-09-30:
Not sure; I would suggest to bundle `libudev` inside the AppImage.

workinghard, 2019-09-30:
@vojtechkral **"all linux distros**" definitely not. But the most recent major ones (Ubuntu/Debian/RH/Suse) have it as a package. During packaging you have to set it as required.

probonopd, 2019-09-30:
PrusaSlicer is shipped as an AppImage for Linux, and AppImages shall not require anything that does not come with the main distributions' default installation.

GarthSnyder, 2019-10-09:
+1. Waiting with bated breath...

ruedli, 2019-10-10:
Yes, indeed: I can see the branch with the 3dconnection feature mentioned
in git... It's coming....

jelmerS2, 2019-11-28:
Any updates on this? :) Would be great to use my space navigator here

Kachidoki2807, 2019-11-28:
According to this comment : https://github.com/prusa3d/PrusaSlicer/issues/3230#issuecomment-558584424, it should be supported on the next release.

Boby71, 2019-11-30:
+1 - thank you!

aderusha, 2019-12-12:
The 2.1.1 release is out now, but I don't see any option to enable 3DConnexion devices nor does it appear to do anything with my device.

lukasmatena, 2019-12-12:
2.1.1 is a minor update of 2.1.0. Most of the new features will be released with 2.2.0.

ruedli, 2019-12-26:
Sooo happy with the first signs of implementation in 2.2.0 Alpha. I feel the motions are not quite all aligned with normal use? I miss the zoom options e.g. The #dconnecxion Trainer, which was installed as part of the driver pack for windows 10, gives quite a good overview of all movements: see 
[image]

All in all there are 6 disitinct (2-directional) movements comming from the device: 3 times panning along x, y, and z and 3 times tilting along x, y and z. Note that the y and z are swapped compared to what a Prusa bed would consider y and z.

Buzzz57, 2020-01-07:
It's great to have the 3Dconnexion mouse implemented. However, it seems like all the movements are the opposite to the "normal" ones and one can't change the behaviour even by going in the 3dconnexion settings pane.

bubnikv, 2020-03-20:
I am closing this one,
the following will be kept open to collect feedback on your preferred configuration of axes.
https://github.com/prusa3d/PrusaSlicer/issues/3852

Please put the screenshots of your 3Dmouse configuration there.

closed 2020-03-20 by bubnikv