prusa3d/PrusaSlicer #1354: multipart - bottom solid layers setting

atxkev, 2018-10-26:
### Version
1.41.1-rc+

### Operating system type + version
MacOS 10.13.6

### Behavior

I have a multipart object. When adding custom part settings to the top most part it seems to ignore Bottom solid layers. You can see in the screen shot it will change the top layer (0 currently to see through) but the bottom layer is missing.

https://www.dropbox.com/s/zr5yq2vyfnr9vhi/multipart-bottomlayer.png?dl=0

no

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

AbeFM, 2018-10-27:
I got the same thing - no bottom and no top if the part rests against another.

It does not matter if "Clip multi-part objects" is on.

[image]

bubnikv, 2019-01-29:
I think you need to enable "Interface shells" at the Print settings -> Multiple Extruders to get what you want.

bubnikv, 2019-04-13:
no response, closing

closed 2019-04-13 by bubnikv

AbeFM, 2019-04-25:
Yes, that seems to have worked. Thanks!